FidoCadJ/FidoCadJ #10: Header ad Footer are needed

MigthyThor, 2015-08-19:
For the sake of formality, content that is reapeated on all pages must be in the Header, Navigation (sidebar menu that we don't need) and Footer section. I suppose we need to move the "titlepage" in an Header and "license" in a Footer.

DarwinNE, 2015-08-19:
Hi MigthyThor,
  I am not sure I understand what you mean.
Are you suggesting a formal change, in the sense that everything which belongs, let's say, to the title should be put in a separated "div" called titlepage? 
In other words, it is a matter of code organization or you are proposing a change in the visual rendering of the page? Can you show me an example of what you mean?

Cheers,
D.

MigthyThor, 2015-08-19:
The visual rendering will remain the same, at least just the license section will be a little different.

The reason why is to organize the page code around the core content. Other stuffs is important, but are all the same in any page you visiting.

Usually you want that the search engine understand immediatly what is the content, and what is the header, navigation and footer of the page. For this reason, I'm thinking to separate formally in the code.

All that remains, of course, in the body element. 

It's just a best practice and you can read more about here: https://css-tricks.com/snippets/html/html5-page-structure/

Cheers.

DarwinNE, 2015-08-19:
Hi,
  now I understand better what you mean.
It is a great idea: if you want to implement it, go on!

Cheers,
D.

DarwinNE, 2015-08-20:
I see you implemented this Issue in the pull request. Thanks!

I am closing it now.

closed 2015-08-20 by DarwinNE