FidoCadJ/FidoCadJ #11: Help needed for updating Linux install instructions

DarwinNE, 2015-08-19:
Hi,
  I need some help to upgrade the Linux install instructions on the English user manual.
They were written by Roby IZ1CYN and are contained in appendix A.2.
Minimum version is Java 7 now and the download locations to be given to wget should be updated to GitHub.
I can modify the document by myself, but I need someone using Linux to check everything.

Cheers,

D.

Stemby, 2015-10-16:
Hi,
I'm a GNU/Linux user, so I would be glad to check everything. Where can I find the updated documentation?

Thank you!

DarwinNE, 2015-10-16:
Hi Stemby,
  I mean in the English manual: 

https://github.com/DarwinNE/FidoCadJ/releases/download/v0.24.5/manual_en.pdf

I think Appendix A.2 should be reviewed (page 56 of the PDF file, 46 of the printed document).
The current release of the file (LaTeX source) is here: https://github.com/DarwinNE/FidoCadJ/blob/master/manual/manual_en.tex
lines 1844-1960

Thank you so much!
D.

Stemby, 2015-10-16:
If minimum version is Java 7 I think you should modify lines 1847-1849.

I found a typo at 1852: "higly" → "highly".

Here is the right output for lines 1864-1880:
```
--2015-10-16 17:57:52--  https://github.com/DarwinNE/FidoCadJ/releases/download/v0.24.5/fidocadj.jar
Resolving github.com (github.com)... 192.30.252.128
Connecting to github.com (github.com)|192.30.252.128|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-cloud.s3.amazonaws.com/releases/39710651/46607e54-3380-11e5-90a8-1b00783bfcf1.jar?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIMWPLRQEC4XCWWPA%2F20151016%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20151016T155753Z&X-Amz-Expires=300&X-Amz-Signature=f6ba4460438c31cda19a42941aa5cdcae8b30c34bd7744774b3755e859b54bd3&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dfidocadj.jar&response-content-type=application%2Foctet-stream [following]
--2015-10-16 17:57:53--  https://github-cloud.s3.amazonaws.com/releases/39710651/46607e54-3380-11e5-90a8-1b00783bfcf1.jar?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIMWPLRQEC4XCWWPA/20151016/us-east-1/s3/aws4_request&X-Amz-Date=20151016T155753Z&X-Amz-Expires=300&X-Amz-Signature=f6ba4460438c31cda19a42941aa5cdcae8b30c34bd7744774b3755e859b54bd3&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment;%20filename=fidocadj.jar&response-content-type=application/octet-stream
Resolving github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)... 54.231.1.241
Connecting to github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)|54.231.1.241|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 639485 (624K) [application/octet-stream]
Saving to: 'fidocadj.jar'

fidocadj.jar        100%[=====================>] 624.50K   680KB/s   in 0.9s   

2015-10-16 17:57:55 (680 KB/s) - 'fidocadj.jar' saved [639485/639485]
$
```

Gwget has been removed from Ubuntu since 2011:

https://bugs.launchpad.net/ubuntu/+source/gwget2/+bug/830427

So I think you should modify line 1930.

In closing I think there is a missing paragraph: how to install FidoCadJ from source.

Ciao!

DarwinNE, 2015-10-16:
Hi, Stemby,
  would you like to modify the code by yourself and do a pull request, so I can integrate your corrections?
You may add your name next to "Roby IZ1CYN" if you want to.

Cheers,

D.

DarwinNE, 2020-06-06:
I am currently updating the manual. I will add a paragraph about compiling FidoCadJ from sources.

DarwinNE, 2020-06-13:
Information has been added to the manual, that should now be reasonably up-to-date.

closed 2020-06-13 by DarwinNE