FidoCadJ/FidoCadJ #13: Download button in the pages

DarwinNE, 2015-08-19:
Probably, we can complement the download page of the website:

with a download button in the homepage. It should recognize which may be the most appropriate version of the program it can be downloaded. I.e. the dmg file for MacOSX, the Windows installer for Windows and so on.

In the past, I wrote something similar in Javascript:
```
<script language="JavaScript1.2" type="text/javascript">
function selectDownload() 
{
	if (navigator.appVersion.indexOf("Mac")!=-1) {
		document.write("<a href='https://sourceforge.net/projects/fidocadj/files/FidoCadJ_MacOSX.dmg/download'><img src='elettronica/fidocadj/icona_download_macosx.png' alt='FidoCadJ icon'><\/a>");
	} else if(navigator.appVersion.indexOf("Win")!=-1) {
		document.write("<a href='http://sourceforge.net/projects/fidocadj/files/FidoCadJ_Windows.msi/download'><img src='elettronica/fidocadj/icona_download_win_installer.png' alt='FidoCadJ icon'><\/a>");
	} else {
		document.write("<a href='https://sourceforge.net/projects/fidocadj/files/fidocadj.jar/download'><img src='elettronica/fidocadj/icona_download_win_linux.png' alt='FidoCadJ icon'><\/a>");
	}
}
</script>
```
The links are quite old, they must be updated to those here on GitHub.

I hope it is useful.

Cheers,

D.

Here are the icons I used at the time (but they are not beautiful). Probably it is better to use only one icon and adapt the text.
[image]
[image]
[image]

MigthyThor, 2015-08-20:
A self reconition of user SO in order to suggest the better version: may be interesting to implement. I'll think about!

DarwinNE, 2015-08-20:
This is what SourceForge used to do, so at a certain point, I tried to do the same in my old pages.

MigthyThor, 2015-09-18:
It works well! Here another version but I think your is better:
http://codepen.io/anon/pen/XmdOvX

We don't need version of OS. Perhaps we can just suggest the Android version when the user is from mobile, if you like.

DarwinNE, 2015-09-18:
The code works, but seems to me quite complex. Do we need all this complexity?

MigthyThor, 2015-09-19:
It cames handly if you need other details, like OS version, but as I wrote, your code is simpler and better for our purpose. I'll use your code. Can you try this condition please? I don't have an android device here:
navigator.appVersion.indexOf("Android")

DarwinNE, 2015-09-20:
Of course!
I do not know how to use codepen, can you please try to put the condition in a code I can test easily with my tablet and my phone?

MigthyThor, 2015-09-20:
Here you can: http://codepen.io/anon/pen/RWRRye

DarwinNE, 2015-09-20:
I tried with my tablet. It works, have a look at the screenshot!
[image]

MigthyThor, 2015-09-20:
Good! We will be able to suggest the android version of FidoCadJ if somebody comes to the site on an android device. Thank you for the test!

DarwinNE, 2015-09-20:
Yes, it seems reliable, I tried also with my smartphone and it works.

MigthyThor, 2015-09-21:
Before I keep working on it, I would have your opinion of current design sketch. I'm not using logos of OS becouse may generate a little bit of trouble in an non expert user (windows has many logos) ad we have not trasparent background logo, I can produce if you are in love with that idea. Let me know what you think about. (the "windows" text will change on differente OS)

[image]

DarwinNE, 2015-09-21:
Not too bad!
I agree. I do not think putting the operating system logo would be useful there.
Just keep the "Download" page as is (the "I need another version of FidoCadJ" must point there), maybe it would be nice to have the version in the button (so something like "FidoCadJ 0.24.5"), but then it would be yet another place where the code has to be changed when a new version has to be distributed. If the text is too long, maybe you can just write "DOWNLOAD FidoCadJ 0.24.5 for Windows".

Thumbs up! Maybe try to experiment with a different color such as green (pick up the one which is in FidoCadJ's icon). 

Cheers,

D.

MigthyThor, 2015-09-21:
>Just keep the "Download" page as is (the "I need another version of FidoCadJ" must point there) 

Of course! With this kind of button, we will need the old download page too. Because something can go wrong with JS code and the download would be unavailable.

>maybe it would be nice to have the version in the button (so something like "FidoCadJ 0.24.5"), but then it would be yet another place where the code has to be changed when a new version has to be distributed. 

It's upon you. I think folks that approach the site from a forum, don't care about version and if they download another time the same version they already have, no problem at all. It's my opinion keep the focus on the very basic user. But I will do what you decide it's right.

>Thumbs up! Maybe try to experiment with a different color such as green (pick up the one which is in FidoCadJ's icon).

I've tried but results wasn't nice and clean. Tomorrow I'll play with colors and post here the results.

Bye!

MigthyThor, 2015-09-24:
Here some versions, I used the color picked from the official FidoCadJ logo:
Darker
[image]
Brighter
[image]
with orange (bleah!)
[image]

If you want to play by yourself, use a CSS button generator website, like this:
http://www.bestcssbuttongenerator.com/

when you find your ideal button, just paste here the CSS code.

Cheers!

DarwinNE, 2015-09-25:
Hi, I prefer the "brighter" version and another friend (DirtyDeeds) also prefers it!

DarwinNE, 2015-09-29:
Thanks for your pull request, which I just merged.

closed 2015-09-29 by DarwinNE