FidoCadJ/FidoCadJ #16: More example

MigthyThor, 2015-08-20:
Can I suggest to add more examples in the examples page to show the FidoCadJ power better?

DarwinNE, 2015-08-20:
Good idea. Maybe we can ask to someone on ElectroYou?
Moreover,  it would be probably nice to provide some newer screenshots.

DanteCpp, 2016-08-30:
I share my screenshot of FidoCadJ under Ubuntu, hoping it will be useful.

[image] 

Sketch from an EY [article](http://www.electroyou.it/dantecpp/wiki/mutuo-accoppiamento-tra-induttori-in-serie).

DanteCpp, 2016-08-30:
Dears @DarwinNE and @MigthyThor,

actually, I proposed it as a screenshot, because, the only Linux screenshot on the website is really old. 
Anyway you may exploit it as you prefer. 

Cheers,
Dante.

DarwinNE, 2016-08-31:
Hi @DanteCpp 
maybe we can update the old screenshot here:

http://darwinne.github.io/FidoCadJ/scrn.html

and then also put the drawing as an example here:

http://darwinne.github.io/FidoCadJ/examples.html

Thanks again,
D.