FidoCadJ/FidoCadJ #17: Appearance of links in the question list of the FAQ pages

DarwinNE, 2015-08-20:
Hi,
  the style of the links in the list of the questions treated in http://fidocadj.sourceforge.net/faq.html is not the same as the links appearing elsewhere. In fact, they are underlined, whereas elsewhere they appear orange or green depending wether they have been clicked or not.

I also noticed that there is a margin problem when viewing that list on my cellular phone (Chrome browser). The bullets are very close to the vertical orange line. I do not have the same effect when viewing the page with my laptop (Safari browser).

I will include a screenshot.

D.

DarwinNE, 2015-08-20:
Here is the screenshot. I noticed that the margin is incorrect only when the phone is oriented vertically.

[image]

MigthyThor, 2015-08-20:
I'm on it!

MigthyThor, 2015-08-20:
Done! I hope. Sending the pull request just now! 

However, little differences depending on mobile browsers and smartphones displays resolutions are normal. I'm not able to fix all of that without a framework. 

May be others able to do, but sorry, I'm not.

bye!

DarwinNE, 2015-08-20:
I think the result is quite good now! 

Just a detail: the "top" link is rendered there in a very small font. Would it be possible to increase its size?

[image]

MigthyThor, 2015-08-21:
Now is done! We are going to find many of this problems without using a framework.

Actually, this one was couse the SMARTphones browsers tend to decide what is relevant content and what isn't. For example if you stay on the right...mhhh: NOT RELEVANT! :-)

DarwinNE, 2015-08-21:
Hi,
  I merged your pull request. 
Now the appearance is very good on my smartphone.

Thank you so much!

Cheers
D.

closed 2015-08-21 by DarwinNE