FidoCadJ/FidoCadJ #24: Simplify layer management

DarwinNE, 2015-08-23:
Hi,
the layer management is quite complex. To show or hide a layer you need to do many steps. Often you need to hide one or more layer just to check the back layer.
I suggest to add a lateral menu for layer selection and hide/edit.

[image]

From https://sourceforge.net/p/fidocadj/feature-requests/9/

paserra, 2019-04-30:
Hi,
for layer management I suggest the simple implementation of two new functions just inserting two (more) buttons in the toolbar (with two corresponding keyboard commands):

Button 1) Hide/Show current selected layer (selected in the toolbat dropbox list)
Button 2) Hide/Show all other layers (selected in the Layer Dialog Management accessible with CTRL-L)

In this way it will be possible to easy manage vias and tracks mainly between copper and components layers in SMD projects.
Cheers,
Pier Andrea.

DarwinNE, 2024-08-16:
@manufino has addressed this issue in a recent pull request #240 
The chosen solution is to modify the combo box to include a visibility icon (clickable) and letting to choose the layer color.

manufino, 2024-08-22:
@DarwinNE, You can close this one for me as well.

closed 2024-08-28 by DarwinNE