FidoCadJ/FidoCadJ #28: Dutch translation needed for UI

DarwinNE, 2015-08-23:
The Dutch translation of the user interface has been somewhat left behind for a while.

Is there someone wanting to give a hand with it?

Here is the file to work on: https://github.com/DarwinNE/FidoCadJ/blob/master/bin/MessagesBundle_nl.properties

Have a look at the end of the file, the untranslated resources are in English.

DarwinNE, 2015-10-13:
I just added some resources, due to new interface elements (printing margins).

DarwinNE, 2015-12-07:
I recently added some resources. Is someone willing to translate them?

DarwinNE, 2016-10-02:
Hi!
I changed the resource 
   
` SpaceKey = The key must not contain spaces or "."`

by adding the 'or "."'

Would be possible to review the translation, please?
Cheers,
D.

lucaslampe, 2016-10-11:
DarwinNE/FidoCadJ#124

DarwinNE, 2017-03-06:
Hi!
Implementing Issue #99 and other things required some new interface elements:
~~~~
Libs=Libs
Attach_image_menu=Image as background
Image_file_attach=Image file to attach:
Image_resolution=Image typographical resolution (dpi):
Can_not_attach_image=Image file could not be attached.
Top_left_coords=X and Y coords of the left topmost corner:
No_img=No image
Image_size_mm=Image width and height in mm:
~~~~

Would be possible to translate them?
Thank in advance!
D.

DarwinNE, 2019-05-11:
I added the following resource while implementing issue #154 

~~~~
Split_layers_multiple_files=Split layers in multiple files
~~~~

Would be possible to translate it?
Thanks in advance

D.

DarwinNE, 2020-04-24:
I updated resource `Grid_width` so to fix #165

DarwinNE, 2020-05-08:
Hi!

I just updated the resource file, to add the following things:

~~~~
Copy_as_image=Copy as image
param_hints=Hints
param_greek=Greek
param_misc=Misc
text_hints=Use "_" to introduce an index or "^" to introduce an exponent. End with the opposite command, i.e. "^" to terminate an index and "_" to terminate an exponent. Type "__" to get a simple underline symbol "_" and "^^" to get a simple caret "^". For example: "3^2_+4^2_=5^2"
~~~~

Would it be possible to translate them, please?

Thank you!!!
D.

langestefan, 2020-05-30:
Hi, I am currently working on updating/adding Dutch translations!

DarwinNE, 2020-05-30:
That's awesome, thank you very much!!!

DarwinNE, 2024-08-16:
I added some resources that should be translated, too:

~~~~
inclusive_selection=Inclusive selection
completive_selection=Selection of complete elements
~~~~

They are the description of the different selection strategies discussed in #236

DarwinNE, 2025-12-27:
Dear all,
   we added some resources to the user interface in the bin/MessageBundle.nl file. Would it be possible to translate them in Dutch for the upcoming 0.24.9 release?

~~~~
inclusive_selection=Inclusive selection
completive_selection=Selection of complete elements
Theme_management=Theme and color management
Theme_groupBox=Theme management
Theme_enableSupport=Enable support for themes
FlatLaf_dark=Dark theme
FlatLaf_light=Light theme
FlatLaf_custom=Customize
Colors_management=Color management
Grid_dots_color=Grid color (dots)
Grid_lines_color=Grid color (lines)
Select_color=Selected elements color
Circuit_backgroud=Drawing area background color
Primitive_color=Default drawing color
Select_RL_color=Right-to-left selection rectangle color
Select_LR_color=Left-to-right selection rectangle color
Custom_theme_info=[The custom theme extends one of the two base themes (dark or light)]
alignLeftSelected=Align left
alignRightSelected=Align right
alignTopSelected=Align top
alignBottomSelected=Align bottom
alignHorizontalCenterSelected=Align horizontal center
alignVerticalCenterSelected=Align vertical center
alignmentMenu=Alignment
distributeHorizontallySelected=Distribute horizontally
distributeVerticallySelected=Distribute vertically
GhostPrimitivesFound=Some primitives have been drawn outside the drawing area.
GhostPrimitivesTranslatePrompt=Do you want to translate all the primitives to bring them into the drawing area?
GhostPrimitivesTitle=Hidden primitives found

Zoom_key=Use the CTRL key to zoom
Move=Move
New_versin=A new version of FidoCadJ is available!
Current_version=Current version:
Latest_version=Latest version
Open_webpage=Do you want to open the download page?
Update_Available=Update Available
Please_visit=Please visit:
Open_Browser=Open Browser
~~~~

Thanks in advance,
D.

PR #124 (2016-10-19): Updated Dutch translations
Fixes #28