FidoCadJ/FidoCadJ #35: Improve the Javadoc quality

DarwinNE, 2015-09-18:
The quality of the Javadoc comments in the source code may be improved by checking the grammar, the completeness and by suggesting if there are obscure points.
The links provided should be improved and validated, as well as the global stylistic coherence and so on.

DanteCpp, 2016-08-26:
I got some errors creating the javadocs 

```
make createdoc 1> /dev/null 2> createdoc.txt
```
The first relevant error 

```
src/net/sourceforge/fidocadj/macropicker/model/AbstractMacroTreeNode.java:23: error: bad use of '>'
    If not, see <http://www.gnu.org/licenses/>.
                                         ^
```
I solved it, substituting 

 `see <http://www.gnu.org/licenses/>`

with 

`@see <a href="http://www.gnu.org/licenses/">http://www.gnu.org/licenses/</a>`

The complete error log file [cretedoc.txt](https://github.com/DarwinNE/FidoCadJ/files/439036/cretedoc.txt).

There are other problems related with the com.apple.* packages.

DarwinNE, 2016-08-26:
Hi @DanteCpp 
thank you, great work!

Have you already changed all links to gnu.org in all source files?
If yes, thanks! It must have been a tedious work. Can you please open a pull request, so that it can be included in the project?
Before doing that, check that the tags do not make sort that the line becomes longer than 80 characters (you may run a ./checkstyle script to see if there are issues).
I think that there may be some similar issues in the Android code, if you have time, can you please have a look at that?

Concerning the missing references to the com.apple.* packages, that is quite normal, if you ran the script on Linux. Apple-specific code is contained only in AppleSpecific.java and this class is dynamically loaded by the startup code of FidoCadJ _only_ if it is running on a MacOSX. Since there is no direct instantiation of an object of this class in the code, it is not needed for the compilation. Yet, you have to compile it explicitly if you need it. Have a look at the compile script:

https://github.com/DarwinNE/FidoCadJ/blob/master/dev_tools/compile

where the relevant bit is here:
`if [ "$unamestr" = 'Darwin' ]||[ "$1" = "mac" ]; then
  echo "MacOSX detected: activated AppleSpecific classes."
  javac  -g -Xlint:unchecked -O -sourcepath src -classpath bin -source 1.7 -target 1.7 ./src/net/sourceforge/fidocadj/AppleSpecific.java -d bin
  echo "Copying Quaqua library files in jar/ directory."
  cp OSes/mac/Quaqua/*.* jar
fi `

Probably, the createdoc script should do something similar: do not include the AppleSpecific.java class unless it is run on MacOSX.

Feel free to try to modify the createdoc  script, if you want to try.

Cheers,
D.