FidoCadJ/FidoCadJ #42: German translation needed for UI

DarwinNE, 2015-10-13:
Dear all,
  I just added some resources, due to new interface elements (printing margins).
Is someone willing to translate them in German, please?

Cheers,

D.

DarwinNE, 2015-10-22:
Thanks to enpa, the translation is complete for the moment, I am closing this issue!

DarwinNE, 2016-10-02:
Hi!
I changed the resource 
   
` SpaceKey = The key must not contain spaces or "."`

by adding the 'or "."'

Would be possible to review the translation, please?
Cheers,
D.

DarwinNE, 2017-03-06:
Hi!
Implementing Issue #99 and other things required some new interface elements:
~~~~
Libs=Libs
Attach_image_menu=Image as background
Image_file_attach=Image file to attach:
Image_resolution=Image typographical resolution (dpi):
Can_not_attach_image=Image file could not be attached.
Top_left_coords=X and Y coords of the left topmost corner:
No_img=No image
Image_size_mm=Image width and height in mm:
~~~~

Would be possible to translate them?
Thank in advance!
D.

DarwinNE, 2017-10-03:
Translation is complete thanks to @crud3 
I close this Issue for the moment.

DarwinNE, 2019-05-11:
I added the following resource while implementing issue #154 

~~~~
Split_layers_multiple_files=Split layers in multiple files
~~~~

Would be possible to translate it?
Thanks in advance

D.

crud3, 2019-05-13:
@DarwinNE 
I can translate this. I noticed some other issues in the German translations which I would also fix with my PR if that's okay.

One question regarding the translations: The word "layer" is sometimes translated with "Schicht" and sometimes with "Ebene". Is this intended or should it be unified? If so, do you have any preferences regarding which translation it should be?

DarwinNE, 2019-05-13:
Hi!

Many thanks!

I don't know the difference between "Schicht" and "Ebene" but the German translation has been done by different people and having different versions of the same word does not seem unexpected to me.

 Corrections and revision of the overall translation will be very welcome.
Thanks in advance,
D.

DarwinNE, 2019-05-13:
Hi @crud3 
I merged the PR, thank you for your work.

I think I will release a preliminary version for beta testing (it will be 0.27.8 beta) in a few days if I can solve a small problem I have with file export.

I close this issue for now, but if you spot typos trying to use the program, don't hesitate to reopen it.

DarwinNE, 2020-04-24:
I updated resource `Grid_width` so to fix #165, so to specify that the grid width must be integer.
Would it be possible to check the translation?

DarwinNE, 2020-05-08:
Hi!

I just updated the resource file, to add the following things:

~~~~
Copy_as_image=Copy as image
param_hints=Hints
param_greek=Greek
param_misc=Misc
text_hints=Use "_" to introduce an index or "^" to introduce an exponent. End with the opposite command, i.e. "^" to terminate an index and "_" to terminate an exponent. Type "__" to get a simple underline symbol "_" and "^^" to get a simple caret "^". For example: "3^2_+4^2_=5^2"
~~~~

Would it be possible to translate them, please?

Thank you!!!
D.

crud3, 2020-05-09:
> I updated resource `Grid_width` so to fix #165, so to specify that the grid width must be integer.
> Would it be possible to check the translation?

Yes, "Gitter Breite (integer)" should be understandable. I believe you could also say "Gitterbreite", but it doesn't really matter.

I'll make a PR with some questions for the rest.

DarwinNE, 2020-05-09:
Thank you very much, @crud3

DarwinNE, 2020-05-17:
Hi @crud3 
   I prepared some preliminary versions of FidoCadJ that got me in touch with some people and they suggested that the little text I wrote to explain how to insert sub and superscripts was not very clear. They suggested to be more "direct" in the description.

I also changed the way one can insert a _ and a ^, so I had to change a little the description of the `text_hints` resource. 

Do you think this version would be clearer than the previous one?

~~~~
Subscripts: insert text between the underscore _ and the caret ^; for example _12^ writes 12 as a subscript. Superscripts: insert text between the caret ^ and the underscore _; for example ^56_ writes 56 as a superscript. In the two cases, the last ^ or _ are optional. To print the _ or ^ character, use a backspace, i.e. \_ to obtain _ and \^ to obtain ^. Use two backspaces to obtain a single \. Example: 3^2_+4^2_=5^2 or C_12
~~~~

If yes, would it be possible to translate it in German?

Kind regards,
D.

enpa, 2020-06-08:
Suggestion for **text_hints**
`Benutze "_" um eine tiefgestellte Sequenz oder "^" um eine hochgestellte Sequenz anzufangen. Beende die Sequenz mit dem gegenteiligen Befehl. Gebe "__" ein, um einen einfachen Unterstrich "_" zu erzeugen und "^^" um ein einfaches Caret "^" zu erzeugen. Zum Beispiel: "3^2_+4^2_=5^2"`

Generally speaking, I'd use a less verbose style and avoid the quotation marks. The abovementioned sentence could be shrinked to
`_ startet eine tiefgestellte Sequenz . ^ startet eine hochgestellte Sequenz. Ende der Sequenz mit dem gegenteiligen Befehl. __ erzeugt einen Unterstrich. ^^ erzeugt ein Caret. Beispiel: 3^2_+4^2_=5^2`
It has (in my opinion) a better readability.

If you want to adopt this style, I can go through the rest of the text.

Cheers, enpa

DarwinNE, 2020-06-09:
Hi @enpa 
The `text_hints` resource in English was difficult to write and the first version was awkward.
I can not judge for German, but if it is possible to simplify things and to make them more accessible, go for it!

Cheers,
D.

DarwinNE, 2024-08-16:
I added some resources that should be translated, too:

~~~~
inclusive_selection=Inclusive selection
completive_selection=Selection of complete elements
~~~~

They are the description of the different selection strategies discussed in #236

enpa, 2024-08-24:
I translated "inclusive selection" to "permissive selection" and "completive selection" to "strict selection". Adding a tooltip with a verbose explanation might help, if feasible.

closed 2024-08-24 by DarwinNE

DarwinNE, 2025-12-27:
Dear all,
   we added some resources to the user interface in the bin/MessageBundle.de file. Would it be possible to translate them in German for the upcoming 0.24.9 release?

~~~~
alignLeftSelected=Align left
alignRightSelected=Align right
alignTopSelected=Align top
alignBottomSelected=Align bottom
alignHorizontalCenterSelected=Align horizontal center
alignVerticalCenterSelected=Align vertical center
alignmentMenu=Alignment
distributeHorizontallySelected=Distribute horizontally
distributeVerticallySelected=Distribute vertically
GhostPrimitivesFound=Some primitives have been drawn outside the drawing area.
GhostPrimitivesTranslatePrompt=Do you want to translate all the primitives to bring them into the drawing area?
GhostPrimitivesTitle=Hidden primitives found

Zoom_key=Use the CTRL key to zoom
Move=Move
New_versin=A new version of FidoCadJ is available!
Current_version=Current version:
Latest_version=Latest version
Open_webpage=Do you want to open the download page?
Update_Available=Update Available
Please_visit=Please visit:
Open_Browser=Open Browser
~~~~

Thanks in advance,
D.

PR #187 (2020-05-09): Added German translations
This PR is work in progress

Fixes #42

I am not sure about the `text_hints`, I think the English text may be wrong. If this is about Exponentiation, the proper terms are **base** and **exponent**, not index. See [here](https://en.wikipedia.org/wiki/Exponentiation). But I couldn't check the exact context of this string, so correct me if I'm wrong.

Also, the example seems incorrect in English. The last exponent (`5^2`) is not terminated. Is this on purpose? If not, it must be fixed in both German and English.
commit a01dff9f (2020-05-09): Added German translations

Closes #42