FidoCadJ/FidoCadJ #43: Spanish translation needed for UI

DarwinNE, 2015-10-13:
Dear all,
  I just added some resources, due to new interface elements (printing margins).
Is someone willing to translate them in Spanish, please?

Cheers,

D.

DarwinNE, 2015-10-17:
Hi marclr,
  thanks a lot for your pull request, it is great!
I will have to add a few other resources in the next days, so I am leaving this issue open for the moment, stay tuned!

Cheers,

D.

DarwinNE, 2015-10-22:
Hi, marclr,
  I think I am not needing for more new resources for the moment.
If you want to complete the translation, I will close that issue.

Cheers,

D.

DarwinNE, 2015-12-07:
I recently added some resources. Is someone willing to translate them?

simon-zz, 2016-04-09:
I am going to do it now. :+1:

simon-zz, 2016-04-09:
Davide, I am trying to push it with git, but seems I don't have the permissions to push the code.

simon-zz, 2016-04-09:
I forked the project and I updated on mine repo. You can fork the file from there. :+1:

DarwinNE, 2016-04-10:
Hi @simon-zz 
Thanks for the translation.

Probably, the easiest thing would be that you do a pull request. 
When I accept it, your code will be automagically integrated with mine.

Cheers,
D.

DarwinNE, 2016-04-15:
Thanks @simon-zz for your pull request!

DarwinNE, 2016-10-02:
Hi!
I changed the resource 
   
` SpaceKey = The key must not contain spaces or "."`

by adding the 'or "."'

Would be possible to review the translation, please?
Cheers,
D.

DarwinNE, 2017-03-06:
Hi!
Implementing Issue #99 and other things required some new interface elements:
~~~~
Libs=Libs
Attach_image_menu=Image as background
Image_file_attach=Image file to attach:
Image_resolution=Image typographical resolution (dpi):
Can_not_attach_image=Image file could not be attached.
Top_left_coords=X and Y coords of the left topmost corner:
No_img=No image
Image_size_mm=Image width and height in mm:
~~~~

Would be possible to translate them?
Thank in advance!
D.

simon-zz, 2017-03-08:
Hello Davide,
Yes, I will update with a pull-request.
s.

DarwinNE, 2017-03-08:
Hi @simon-zz 
Your contribution will be much welcomed, as usual!
Thanks!

Davide

DarwinNE, 2019-05-11:
I added the following resource while implementing issue #154 

~~~~
Split_layers_multiple_files=Split layers in multiple files
~~~~

Would be possible to translate it?
Thanks in advance

D.

gvee-s, 2019-05-12:
Hello Davide,
It should be 

`Split_layers_multiple_files=Dividir capas en múltiples ficheros`

I looked for the string in the source code but I did not found it.
s.

DarwinNE, 2019-05-12:
Hi @gvee-s

It's in the file bin/MessageBundle_es.properties:

https://github.com/DarwinNE/FidoCadJ/blob/master/bin/MessagesBundle_es.properties

Can you update it and send a PR?

Cheers,
D.

gvee-s, 2019-05-12:
Sorry @DarwinNE , my bad.
I was looking an older branch.

I will  PR in the next minutes.
Cheers,
s.

DarwinNE, 2019-05-12:
Hi @gvee-s I merged the PR, thank you very much!

I realized I forgot to mention that I also added a few months ago the following resources:

~~~~
res_export=Set resolution in dpi
size_export=Set size in pixels
~~~~

Would you mind translating them, too, please?

Cheers,
D.

gvee-s, 2019-05-12:
Done! :+1: 
Cheers,
s.

DarwinNE, 2019-05-12:
Thank you very much, @gvee-s 

Cheers,
D.

DarwinNE, 2020-04-24:
I updated resource `Grid_width` so to fix #165, so to specify that the grid width must be integer.
Would it be possible to check the translation?

gvee-s, 2020-05-05:
Hello @DarwinNE ,

`Grid_width = Grosor de la cuadricula (número entero):`

Looks perfect. :+1: 
s.

DarwinNE, 2020-05-05:
Thank you @gvee-s 

I will probably add some resources for the version 0.24.8 later, but for the moment I close this issue.

DarwinNE, 2020-05-08:
Hi!

I just updated the resource file, to add the following things:

~~~~
Copy_as_image=Copy as image
param_hints=Hints
param_greek=Greek
param_misc=Misc
text_hints=Use "_" to introduce an index or "^" to introduce an exponent. End with the opposite command, i.e. "^" to terminate an index and "_" to terminate an exponent. Type "__" to get a simple underline symbol "_" and "^^" to get a simple caret "^". For example: "3^2_+4^2_=5^2"
~~~~

Would it be possible to translate them, please?

Thank you!!!
D.

DarwinNE, 2020-06-12:
Thanks @gvee-s as usual for the pull request. I can close this issue now.

DarwinNE, 2024-08-16:
I added some resources that should be translated, too:

~~~~
inclusive_selection=Inclusive selection
completive_selection=Selection of complete elements
~~~~

They are the description of the different selection strategies discussed in #236

simo-zz, 2024-12-11:
Hello Davide, 

Let me check during these days.

Thanks,
Simon

DarwinNE, 2025-12-27:
Dear all,
   we added some resources to the user interface in the bin/MessageBundle.es file. Would it be possible to translate them in Spanish for the upcoming 0.24.9 release?

~~~~
inclusive_selection=Inclusive selection
completive_selection=Selection of complete elements
Theme_management=Theme and color management
Theme_groupBox=Theme management
Theme_enableSupport=Enable support for themes
FlatLaf_dark=Dark theme
FlatLaf_light=Light theme
FlatLaf_custom=Customize
Colors_management=Color management
Grid_dots_color=Grid color (dots)
Grid_lines_color=Grid color (lines)
Select_color=Selected elements color
Circuit_backgroud=Drawing area background color
Primitive_color=Default drawing color
Select_RL_color=Right-to-left selection rectangle color
Select_LR_color=Left-to-right selection rectangle color
Custom_theme_info=[The custom theme extends one of the two base themes (dark or light)]
alignLeftSelected=Align left
alignRightSelected=Align right
alignTopSelected=Align top
alignBottomSelected=Align bottom
alignHorizontalCenterSelected=Align horizontal center
alignVerticalCenterSelected=Align vertical center
alignmentMenu=Alignment
distributeHorizontallySelected=Distribute horizontally
distributeVerticallySelected=Distribute vertically
GhostPrimitivesFound=Some primitives have been drawn outside the drawing area.
GhostPrimitivesTranslatePrompt=Do you want to translate all the primitives to bring them into the drawing area?
GhostPrimitivesTitle=Hidden primitives found

Zoom_key=Use the CTRL key to zoom
Move=Move
New_versin=A new version of FidoCadJ is available!
Current_version=Current version:
Latest_version=Latest version
Open_webpage=Do you want to open the download page?
Update_Available=Update Available
Please_visit=Please visit:
Open_Browser=Open Browser
~~~~

Thanks in advance,
D.

DarwinNE, 2026-01-08:
Hi @gveezz 
Thank you very much for the pull request #278 

Shall we close this issue?

commit a6ba07be (2017-03-13): Merge pull request #136 from simon-zz/master

Updated MessagesBundle_es.properties 11-03-17, fix #43
closed 2026-01-20 by DarwinNE