FidoCadJ/FidoCadJ #51: Unix permissions

Stemby, 2015-10-17:
Unix permissions for all files should be checked and made uniform.

DarwinNE, 2015-10-18:
Hi,
I agree, probably the best permissions for source files may be 644 and 755 for directories. One may check and correct that.
BTW, it is quite easy to mess with permissions when working with different operating systems.
Once the problem is solved, would it be a good idea to add a test for consistency?
Currently, there are Checkstyle tests run on the source code (checkstyle.sh), can it be expanded to check the permissions?

Cheers,
D.

DarwinNE, 2015-11-15:
Hi. That should be correct now. I close the issue, if there are any other leftout, I will reopen it.

closed 2015-11-15 by DarwinNE