FidoCadJ/FidoCadJ #53: Correct warnings issued by Java 1.7

DarwinNE, 2015-10-18:
Hi,
  during the compilation of Java sources, some warnings are issued.
They are mainly due to the older code (Java <1.5) using unchecked casts:

~~~~~~
$ ./rebuild 
MacOSX detected: activated AppleSpecific classes.
src/net/sourceforge/fidocadj/macropicker/model/MacroTreeModel.java:124: warning: [unchecked] unchecked conversion
        libraryNodeMap = new HashMap();
                         ^
  required: HashMap<TreePath,AbstractMacroTreeNode>
  found:    HashMap
src/net/sourceforge/fidocadj/macropicker/model/MacroTreeModel.java:462: warning: [unchecked] unchecked cast
               (HashMap<TreePath,AbstractMacroTreeNode>)libraryNodeMap.clone();
                                                                            ^
  required: HashMap<TreePath,AbstractMacroTreeNode>
  found:    Object
src/net/sourceforge/fidocadj/macropicker/model/AbstractMacroTreeNode.java:118: warning: [unchecked] unchecked conversion
        Collections.sort(childNodes);
                         ^
  required: List<T>
  found:    Vector
  where T is a type-variable:
    T extends Comparable<? super T> declared in method <T>sort(List<T>)
src/net/sourceforge/fidocadj/macropicker/model/AbstractMacroTreeNode.java:118: warning: [unchecked] unchecked method invocation: method sort in class Collections is applied to given types
        Collections.sort(childNodes);
                        ^
  required: List<T>
  found: Vector
  where T is a type-variable:
    T extends Comparable<? super T> declared in method <T>sort(List<T>)
src/net/sourceforge/fidocadj/macropicker/model/MacroTreeModel.java:554: warning: [unchecked] unchecked call to add(E) as a member of the raw type Vector
            childNodes.add(node);
                          ^
  where E is a type-variable:
    E extends Object declared in class Vector
src/net/sourceforge/fidocadj/macropicker/model/MacroTreeModel.java:586: warning: [unchecked] unchecked call to add(E) as a member of the raw type Vector
            childNodes.add(node);
                          ^
  where E is a type-variable:
    E extends Object declared in class Vector
src/net/sourceforge/fidocadj/macropicker/model/MacroTreeModel.java:636: warning: [unchecked] unchecked call to add(E) as a member of the raw type Vector
            childNodes.add(node);
                          ^
  where E is a type-variable:
    E extends Object declared in class Vector
src/net/sourceforge/fidocadj/toolbars/ToolbarZoom.java:217: warning: [unchecked] unchecked cast
        JComboBox<String> source=(JComboBox<String>)evt.getSource();
                                                                 ^
  required: JComboBox<String>
  found:    Object
src/net/sourceforge/fidocadj/dialogs/DialogExport.java:537: warning: [unchecked] unchecked cast
            (JComboBox<String>)(evt.getSource());
                               ^
  required: JComboBox<String>
  found:    Object
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
9 warnings
~~~~~~

We can of course suppress those warnings (the code has been long tested), but I would prefer to issue a more elegant solution.

Any suggestions?

marclr, 2015-10-29:
In my opinion the good way is migrating to 1.7 to have the new features added before suppress the warnings, 9 warnings are not so much to correct it and migrate it

DarwinNE, 2015-10-29:
Hi,
  I agree, but I do not know how to handle those particular warnings...
In other words, how would be the new code?

Cheers,
D.

DarwinNE, 2016-01-25:
There is no more warning issued by the Java compiler. Mission accomplished :-)

closed 2016-01-25 by DarwinNE