FidoCadJ/FidoCadJ #54: README: replacing old links

Stemby, 2015-10-21:
In par. 1.4 there are two links to SF that should be replaced.

DarwinNE, 2015-10-21:
Thanks Stemby, I will correct that soon!

DarwinNE, 2015-10-29:
Since I you are already working with the README files, I think it is a good idea that you correct the file by yourself. Here is what I propose:

~~~~~
If you feel lost, you can have a look at the manuals. Languages different 
from English might be outdated or no more listed:

https://github.com/DarwinNE/FidoCadJ/releases

(search for the pdf files!)
If you need help, you can post a question as a GitHub issue:

https://github.com/DarwinNE/FidoCadJ/issues
~~~~~

The problem is that the Italian translation is outdated and the French version is *really* outdated (see Issues #49 and #50), so I do not think it is a good idea to include them in the last Release (still the 0.24.5), even if they are still available on the old Sourceforge download page. My hope is that someones will give a hand for that in a near future!

commit 36dd7850 (2015-10-29): SF → GitHub; closes #54.
closed 2015-10-29 by DarwinNE