FidoCadJ/FidoCadJ #59: Multi language website for FidoCadJ

MigthyThor, 2015-10-29:
We need to add multi language support for the website. That's my plan to do it:

I'm going to transform the current web site (in html) in a PHP web site. 

PHP web sites don't run on GitHub, but in the future who knows... for now we use it to **generate the html code.**

With a multi language web site made in PHP we will have a common file for the template, and many content file per page as many language we will use. Technically speaking, we separate the content from the viewer.

So, all we have to do, is **run the PHP site on a local server**, for every chang on web site (that aren't so frequently) copy and paste the html source code in the GitHub directory.

I'll try to make **an example** on a piece of html code from page index.html

Now the title of the page is defined by this code:
`<title>FidoCadJ Free Graphical Editor for Electronics and more</title>`

In the PHP version there'll be an include about a language file(for example italian):
`include("languages/italian/index.php");`
that contain the languages definition, the title will change in:
`<title><?php echo $TITLE; ?></title>`

If you will look at the output/source code of the new index.php, will see only the result:
`<title>FidoCadJ Software gratuito per il disegno di circuiti elettronici e altro</title>`

So will have a new structure for the other (not english) languages:
Italian -> http://darwinne.github.io/FidoCadJ/italian/index.html
Spanish -> http://darwinne.github.io/FidoCadJ/spanish/index.html
Klingon -> http://darwinne.github.io/FidoCadJ/klingon/index.html

and so on...

Of course, **I will tell Google wich languages we ar going to serve** and I'll make some beautiful **flags** for the user to choose the language he prefers.

**I will translate the website in italian** and I'll make a **little guide for new translators** about how to edit the language file.

Hope I've been clear enough.

Cheers!

marclr, 2015-10-29:
And why not use angularJS?  I think that github could host the web

MigthyThor, 2015-10-29:
DarwinNE doesn't like libraries, so...neither Jquery

DarwinNE, 2015-10-29:
Indeed, I prefer to handle simple code, when possible.
I already employed PHP in the past for multilanguage websites and it seemed to work correctly. The advantage of PHP is that I understand a little the language (that I do not like very much).
However, the need of running a website server to compile it is quite a PITA :-)

What I like for that solution is that the result is a HTML-only website which has the advantage that it can easily be made very responsive.

If it can help, here is the source code of my homepage, I wrote circa 2006 and it still works. I had to be hyper-attentive about employing the "input" command, since I had an hacker attack in the past, on another website.
~~~~~
<?php

ob_start();

function validateLanguage($lang)
{
    $isValidLanguage['English']=true;
    $isValidLanguage['Francais']=true;
    $isValidLanguage['Italiano']=true;
    
    if($isValidLanguage[$lang]==true) {
        return $lang;
    }else{
        return "";
    }
}

/* Leggi gli argomenti passati tramite l'URL 
   Per compatibilità con Alice, uto HTTP_GET_VARS per avere un codice
   che funziona con il loro vetusto PHP 4.0.1 */
   
if(substr(phpversion(),0,1)<"5")
    $oldphp=true;
else
    $oldphp=false;
    
if($oldphp) {
    $page=$HTTP_GET_VARS["argument"];
    $language=$HTTP_GET_VARS["language"];
    $exp=$HTTP_GET_VARS["exp"];

} else {
    $page=$_GET["argument"];
    $language=$_GET["language"];
    $exp=$_GET["exp"];

}

/* Piccolo test, per la sicurezza */
$language=validateLanguage($language);

/* Se non viene fornito nulla, mostra la home page*/
if($page=="") {
    $page="home.inc";
}
   
/* Se la lingua è specificata espressamente, memorizza l'impostazione
   in un cookie che dura un anno */
     
if($language!="") {
    setcookie("language", $language, time()+365*24*8644);
}   
   
/* Se la lingua non è specificata espressamente, cerca un cookie. 
   Se questo non è presente, prediligi la versione in francese, sempreché 
   questa sia disponibile */
 
 if($language=="") {
    if($oldphp) 
        $ll=$HTTP_COOKIE_VARS["language"];
    else
        $ll=$_COOKIE["language"];
[... 151 lines cut ...]
    $out=str_replace("%TITLE%", $page_title, $out);
    
    if($description!="") 
        $desc = "<meta name=\"description\" content=\"".$description."\">";
    else
        $desc = " ";
        
    $out=str_replace("%DESCRIPTION%", $desc, $out);
    echo str_replace("%LANGUAGE%", $locale, $out);
?>
</HTML>
~~~~~
Retrospectively, I think that the URLs of my webpage are too long. Therefore, I think that employing the standard locale codes for the language would be enough. I am sure that a PHP wizard would find a way to improve it, but it has proven to be acceptable, at least for my small personal webpage.

However, I am not *completely* against employing external libraries. Here are my requirements:
- very small overhead for the download (ex. 30 KiB is OK, 150 KiB not at all)
- clean, small, understandable code
- clean, simple, understandable structure
- no security issues
- no need of spending more time to upgrade the framework than the site

For example, my homepage still runs code written in 2006, since I can not upgrade versions of it every 2 months. /Always think for the long run/.

Cheers,
D.

MigthyThor, 2016-01-11:
Hi,
italian translation is done! Now I have to add a language-navigation system to change languages within the website.

It will be usefull to know how many languages you think will be integrated.

Thanks.

DarwinNE, 2016-01-11:
That's truly great, @MigthyThor!
In the name of the Italian user community, THANKS for the commitment and for your efforts!

Can I have a look at the code? Then when you finish the navigation system, may I propose that you do a pull request so it will be integrated in the current website?

Best regards,

Davide

MigthyThor, 2016-01-18:
Sorry for delay in answer!

You can look at the code in my repository (go_multilanguage branch) I suppose. If not possible, I'll send you the files.

Bye.

DarwinNE, 2016-01-18:
Hi, @MigthyThor 
I could have a look at the go_multilanguage. That's great!
I have only a minor cosmetic remark: I noticed that the tab length seems to be rendered with 8 spaces here on GitHub. For that reason, I have chosen to get rid completely of the tab in FidoCadJ sources (my editor can "detab" very easily the sources). I substituted them with 4 spaces, which is what I employ usually.

Thanks again for your commitment: great work!

Cheers,

Davide

DarwinNE, 2016-04-05:
Hi @MigthyThor
 thanks to your work, the site is now bilingual, in English as well as in Italian.
Great work and thanks for the pull requests!

I think we can now close this Issue.

D.

closed 2016-04-05 by DarwinNE