FidoCadJ/FidoCadJ #71: subscripts

Fedhman, 2015-12-08:
They would be really usefull, expecially to differenciate resistors and capacitors (and so on).

DarwinNE, 2015-12-08:
I agree, and the same is true for superscripts.

I thought it would be reasonable to implement a mini-markup language such as follows:

R_1  for R<sub>1</sub> or I^2 for I<sup>2</sup> and braces for longer stuff such as h_{fe} for h<sub>fe</sub>.

What do you think about it?

Fedhman, 2015-12-08:
As LaTex programming languages. Yeah it would be perfect

DarwinNE, 2015-12-08:
Exactly. I love LaTeX!

DarwinNE, 2020-05-03:
> R_1 for R1 or I^2 for I2 and braces for longer stuff such as h_{fe} for hfe.

After much thought, I decided against it. This would require to implement a sort of a mini-language and I don't really want to do that. Instead, I think a good solution may be to use a very simple convention:

~~~~
^  -> goes towards the exponent
_ -> goes towards the index
~~~~

The meaning of those rules is the following one:

`x^2` will be  interpreted as "x<sup>2</sup>"
`this is a ^test_ where the _rest^ is not important` will yield  "this is a <sup>test</sup> where the <sub>rest</sub> is not important"

In other words, in the second example the _ cancels the effect of ^ and vice versa.

DarwinNE, 2020-05-21:
It has not been very easy, but at the end the implementation of superscripts and subscripts in the vector file export could be done relatively smoothly.

I finally opted for this description:

~~~~
Subscripts: insert text between the underscore _ and the caret ^; for example _12^ writes 12 as a subscript. Superscripts: insert text between the caret ^ and the underscore _; for example ^56_ writes 56 as a superscript. In the two cases, the last ^ or _ are optional. To print the _ or ^ character, use a backspace, i.e. \_ to obtain _ and \^ to obtain ^. Use two backspaces to obtain a single \. Example: 3^2_+4^2_=5^2 or C_12
~~~~

(resource `text_hints` in the English language).

I close this issue now, as the implementation is complete.

closed 2020-05-21 by DarwinNE