FidoCadJ/FidoCadJ #72: Independent distro installer script

DanteCpp, 2015-12-10:
When i run the [script](https://github.com/DarwinNE/FidoCadJ/blob/master/OSes/linux/Distro-Independant/fidocadj-installer-linux.sh) I get

```
dante@dante-laptop:~/Documenti/varie$ sudo ./fidocadj_installer.sh --install
 
   Welcome to the fidocadj_installer script
 
   Cannot reach the SourceForge server!
Abort... 
 
dante@dante-laptop:~/Documenti/varie$
```

it is because the installer still tests the sourgeforce server

```bash
## connection availability test
ping -c1 download.sourceforge.net > /dev/null
if [ $? -ne 0 -a $1 != "--uninstall" ]
then
    echo " "
    echo "   Cannot reach the SourceForge server!"
    echo "Abort... "
    echo " "
    exit 3
fi
```

without this test, the script would try to download the fidocadj.jar file by the following command

```bash
 wget -q -O /usr/share/java/fidocadj/temp.jar http://download.sourceforge.net/fidocadj/files/fidocadj.jar
```

I will fix that, but i need the new fidocadj.jar download link.

Cheers Dante.

DarwinNE, 2015-12-10:
Hi, that is an interesting problem. 
Binaries on GitHub are distributed with the "Releases":

https://github.com/DarwinNE/FidoCadJ/releases

This means that the fidocadj.jar file related to the last stable version (0.24.5 while I am writing) is there:

https://github.com/DarwinNE/FidoCadJ/releases/download/v0.24.5/fidocadj.jar

When the stable "0.24.6" version will be released, the link will be something as follows:

https://github.com/DarwinNE/FidoCadJ/releases/download/v0.24.6/fidocadj.jar

So, the link will change at each version of the software. The script should see which releases are present on GitHub and install the most recent one. Alternatively, it should let the user choose which version he/she wants to install.

Cheers,
D.

DanteCpp, 2015-12-11:
Hi Davide, 

I see at least three ways:
* Include in the compile.sh file a subroutine that change the link in the installer according to the version. (this way keeps simplicity, but the user need to download the new installer for any version).
* Implement in the installer, the research of the newest version. (This make the code more complex).
* Leave user choose the version. (That way in simple and already done)

The installer includes the --restore [version] option:

```
dante@dante-laptop:~/Documenti/varie$ ./fidocadj_installer.sh --help
        ....
 	fidocadj_installer.sh --install      	   installs FidoCadJ
 	fidocadj_installer.sh --uninstall    	   uninstalls FidoCadJ
 	fidocadj_installer.sh --update       	   updates your FidoCadJ jar file
 	fidocadj_installer.sh --restore [VERSION]    restore your FidoCadJ jar file to an old version
        ...
```
I prefer the second way, but for make it more simple i would include a given file that contains the current stable version and other important info. 
So the installer script can query directly this file.

I hope I've been clear.

Cheers, Dante.

DarwinNE, 2015-12-11:
Hi,
  the second solution is preferable.
Having a file containing information about the most recent stable version seems potentially troublesome to me, since it requires to change it each time a new version comes out. This is quite error prone in my opinion, since the release of a new version is already a complex process.
So, here is what I propose:
1 - let the user choose which version he wants to install
2 - GitHub has a powerful API system. There may probable be a way to retrieve the list of all the available releases.
3 - The installer chooses the most recent release, flagged as "stable".

Those three points should be of a growing complexity and each level is self-sufficient (i.e. one can implement 1. without 2. and 3.) What do you think about it, Dante?

Cheers,

Davide

Stemby, 2015-12-12:
Hi,
actually I'm not interested in this kind of scripts (I won't use it), so I can't help you, sorry.

Currently I compile by hand from source; in future I hope to use an official package taken from the official repositories of my distribution and installed via the standard tools provided by the distro.

In any case I agree, the second way seems to be the better one; this URI could be a good start:

https://github.com/DarwinNE/FidoCadJ/releases/latest

Ciao!

DarwinNE, 2015-12-12:
Hi @Stemby 
  this URL is exactly what it should be used!
Even for the website, I may be asking to @MigthyThor what he thinks about using it even for the download page, instead of specifying which version has to be downloaded, and to pack the version number shown in the text on a variable, easy to change somewhere.
I am really trying to see if the release operation of a new version can be made as simple as possible!
Cheers,
D.

DarwinNE, 2016-01-05:
Hi @DanteCpp,
  were you able to include the link given by @Stemby in the script?

Cheers,
D.

DanteCpp, 2016-08-26:
Solved by implementation of the `make` automated build system. 

## How to install FidoCadJ on Linux
In the following instruction, you have to substitute X.XX.X with the wanted FidoCadJ version.

-  Download the source code 
` wget https://github.com/DarwinNE/FidoCadJ/archive/vX.XX.X.tar.gz`

- Unpack the tarball 
`tar -zxvf FidoCadJ-X.XX.X.tar.gz`

- Go in the source directory, compile and install 
`cd FidoCadJ-X.XX.X/`
`make`
`sudo make install`

closed 2016-08-26 by DanteCpp

DarwinNE, 2016-08-26:
I agree with @DanteCpp, the new procedure is standard for Linux systems and is clearly a better solution.

@DanteCpp proposed the new scripts in this pull request: #106