FidoCadJ/FidoCadJ #75: Compatibility mode with FidoCAD

wall87, 2016-01-29:
Hi
When I'm going to draw the curve button is freezes and not selectable, even once draw a line will not let me choose if I want an arrow.

DarwinNE, 2016-01-29:
Hi @wall87,
  this is normal, since the original "FidoCAD per Windows" (not developed since 2001) did not let to do such kind of things. The "compatibility mode" is meant for those who want to obtain a drawing compatible with that very old program.

However, there is a usability issue, which in the past caused a lot of troubles for users: people activate that mode and then forget about it wandering after a wile why certain functions are not available.

For that reason, from version 0.24.6 (we are working on that right now), the "FidoCAD compatibility mode" will not be conserved each time FidoCadJ is restarted.

Kind regards,
D.

DarwinNE, 2016-01-29:
Here is a screenshot of FidoCadJ running with the 'FidoCAD' compatibility mode active (MacOSX 10.11.2, French locale):
[image]

closed 2016-03-25 by DarwinNE