FidoCadJ/FidoCadJ #77: Text with negative x-dimension (or y-dimension) is hidden

Max2433BO, 2016-02-03:
Hi DarwinNE,

Normally, if you enter a non numeric on x or y dimension (in text parameter), you see the warning:

> **Non riesco a leggere uno dei numeri introdotti**
(Italian version)

and to continue you have to correct the wrong input.

Instead, if you enter a negative number, input is accepted but nothing is displayed display area, and it looks like processed command.

Example: generated code with negative x-dimension input

[FIDOCAD]
FJC B 0.5
TY 60 50 4 -3 0 0 0 * String

Bye, Max

DarwinNE, 2016-03-28:
Hi @Max2433BO 
I am trying to correct this behavior, but how should FidoCadJ correct it? Should it just eliminate the text element? Should it convert the negative value into a positive one? Should show an error message?
Cheers,
D.

DarwinNE, 2016-03-29:
I decided that if a text has a size <1 then its size is put equal to 1.
I also introduced a maximum size of 2000 for the size, excesses on the other side are not giving problems.

closed 2016-03-29 by DarwinNE