FidoCadJ/FidoCadJ #78: Reversed arrow

Max2433BO, 2016-02-03:
Hi, DarwinNE

I read english manual (version 8/19/2015), at the 2.7 "Arrow and stroke styles" but I have not found any trace of this feature:

In a bezier (or normal) curve, if you set on "display arrow" (at the start or at the end), and you set "length tip" with a negative number, the arrow is show reversed.

[image]

However, with the line this "function" does not work.

This is the code only for curves:

[FIDOCAD]
FJC B 0.5
CV 0 15 5 40 35 55 20 0
FCJ 2 0 3 2 0 0
BE 85 20 115 35 130 15 150 30 0
FCJ 2 0 3 2 0 0
CV 0 15 45 40 75 55 60 0
FCJ 2 0 -3 2 0 0
BE 85 65 115 80 130 60 150 75 0
FCJ 2 0 -3 2 0 0

Bye, Max

DarwinNE, 2016-02-15:
Hi @Max2433BO 
  at the beginning, I thought it was a bug, but then I realized that it may be interesting to document this behavior and make sort that it happens the same thing with the lines.

Cheers,
D.

Max2433BO, 2016-03-06:
Good idea!!!

Bye, Max

DarwinNE, 2016-03-29:
Everything should have been taken into account and updated. Negative length for arrows is now accepted and documented.

closed 2016-03-29 by DarwinNE