FidoCadJ/FidoCadJ #79: change line width

wall87, 2016-03-07:
Hi DarwiNE
can you associate with each a different thickness line?

DarwinNE, 2016-03-07:
Yes you can by using the PCB line primitive and not the "line" one.
However, the width must be integer for the moment.

FidoCadJ in version 0.24.5 accepts and parses non integer values for the "line width" parameter, rounding them to integers. Starting from version 0.24.6, we will make sort that a non-integer value is taken into account and employed in the drawings. In other words, the current version of FidoCadJ is already accepting the new file format :-)

Doing such, when the new version will be released, people still employing the older version of FidoCadJ can render the drawing albeit with a rounded PCB line widths.

DarwinNE, 2016-03-22:
It works now. I still have to check what happens with the exported drawings.

DarwinNE, 2016-04-07:
I tried to export non integer pcb track sizes in different vector formats and it seems to work. 
I therefore close that Issue.

closed 2016-04-07 by DarwinNE