FidoCadJ/FidoCadJ #80: Selection of ellipse or circle issue

DarwinNE, 2016-03-24:
Dear Programmer, I have FidoCADj v0.24.6 Alpha installed in my PC, with Windows XP - Java v1.7, and I verify an issue in this scheme:
[FIDOCAD]
FJC C 1.5
FJC A 0.35
RV 132 14 389 153 0
FCJ 3 0
PV 204 30 204 66 197 66 197 96 203 96 203 132 207 132 209 127 215 127 216 132 220 132 220 96 226 96 226 66 219 66 219 30 215 30 214 35 208 35 207 30 204 30 204 30 0
LI 197 76 226 76 0
FCJ 0 0 3 2 1 0
TY 234 78 4 3 0 0 0 Arial D4
TY 161 77 4 3 0 0 0 Arial D2
EV 275 37 367 129 0
FCJ 1 0
TY 138 76 4 3 0 0 0 Arial D1
LI 197 86 226 86 0
FCJ 0 0 3 2 1 0
TY 224 47 4 3 0 0 0 Arial G3
EV 269 31 372 135 0
TY 240 30 4 3 0 0 0 Arial G2
TY 207 17 4 3 0 0 0 Arial G1
TY 208 141 4 3 0 0 0 Arial S1
TY 180 77 4 3 0 0 0 Arial D3
EV 305 65 337 97 0
EV 316 76 326 86 0
LI 318 85 318 90 0
LI 318 90 324 90 0
LI 324 90 324 85 0
LI 197 91 226 91 0
FCJ 0 0 3 2 1 0
LI 184 66 197 66 2
FCJ 0 0 3 2 1 0
LI 184 96 197 96 2
FCJ 0 0 3 2 1 0
LI 203 142 203 132 2
FCJ 0 0 3 2 1 0
LI 214 35 230 35 2
FCJ 0 0 3 2 1 0
LI 164 35 208 35 2
FCJ 0 0 3 2 1 0
LI 142 30 204 30 2
FCJ 0 0 3 2 1 0
LI 235 86 226 86 2
FCJ 0 0 3 2 1 0
LI 215 30 215 21 2
FCJ 0 0 3 2 1 0
LI 219 30 230 30 2
FCJ 0 0 3 2 1 0
LI 207 30 207 21 2
FCJ 0 0 3 2 1 0
LI 214 44 214 35 2
FCJ 0 0 3 2 1 0
LI 164 127 209 127 2
FCJ 0 0 3 2 1 0
LI 208 44 208 35 2
FCJ 0 0 3 2 1 0
LI 220 142 220 132 2
FCJ 0 0 3 2 1 0
LI 141 132 203 132 2
FCJ 0 0 3 2 1 0
LI 235 76 226 76 2
FCJ 0 0 3 2 1 0
LI 383 81 258 81 4
FCJ 0 0 3 2 4 0
LI 321 19 321 144 4
FCJ 0 0 3 2 4 0
LI 211 28 212 135 4
FCJ 0 0 3 2 4 0
LI 195 81 228 81 4
FCJ 0 0 3 2 4 0
LI 207 25 215 25 9
FCJ 3 3 3 2 0 0
LI 223 47 215 42 9
FCJ 2 2 3 2 0 0
LI 230 76 230 86 9
FCJ 3 3 3 2 0 0
LI 203 138 220 138 9
FCJ 3 3 3 2 0 0
LI 169 35 169 127 9
FCJ 3 3 3 2 0 0
LI 146 30 146 132 9
FCJ 3 3 3 2 0 0
LI 237 32 231 32 9
FCJ 2 2 3 2 0 0
LI 189 66 189 96 9
FCJ 3 3 3 2 0 0
When I select the circle on the right side of scheme, sometime the circle is impossible to select, the issue is again if I zooming the schematic.
This problem is also in the v0.24.5 (stable version) and I verified the stranger issue in an other PC with OS Windows 8 and Java v1.8
Dear DarwinNE is possible verify if the stranger problem is again in your OS configuration?
An other problem is in drawing any circle from the ellipse tool and CTRL key.
If I select the tool ELLIPSE and driwing an circle with CTRL, the circle is driwed all ok, but if I select the previous circle drawing, is impossible the modify the diameter of the circle selected because is transformed automatically in ellipse and is impossible the use of CTRL key for reproduce an other circle from the first.
In other word is impossible correct an circle in the diameter because the CTRL key not influence the drawing in selected mode.
Many thanks for your interest to my problem.

Transferred from: https://sourceforge.net/p/fidocadj/bugs/65/

DarwinNE, 2016-03-24:
Hi and thanks for the report, can you please try with FidoCadJ 0.24.6 beta?

Cipollino, 2016-03-25:
Dear DarwinNE, I use the v0.24.6 Alpha now (I don't retrive the Beta version on internet).
Please send to me the address of dowload for your last version od FidoCADj and I test if the issue is again in his last version.

Many thanks for your interest to my problem.

Bye for now.

Cipollino, 2016-03-25:
Dear DarwinNE, I download now the v0.24.6 beta and I try it, but both the problems is present again.

DarwinNE, 2016-03-25:
Hi Cipollino,
  thanks for your report. Thumbs up!

I try to see if I can reproduce the problem.

Cheers,
D.

Cipollino, 2016-03-25:
Dear DarwinNE, for first, many thanks for your interest to my problem.
For your information, I tested the v0.24.6 beta on 3 different PC with:
OS Windows XP (with HD monitor 16:9);
OS Windows 8 (with portable  13" monitor HD 16:9);
OS Windows XP (with 19" LCD square monitor)

All 3 PC have different screen resolution and both 2 problems is the same. 
Good Ester Holyday's for you and your Family.

DarwinNE, 2016-03-25:
Hello,
  I think I could reproduce the problem and I patched it.
Can you please have a look if it is ok in the future? I plan releasing a new preliminary version (v0.24.6 gamma) next week and it will include the correction.

Cheers,
D.

Cipollino, 2016-03-26:
I'm in waiting.

Good Ester Holydays

DarwinNE, 2016-03-29:
Hi Cipollino,
  can you please have a look at FidoCadJ 0.24.6 gamma?

https://github.com/DarwinNE/FidoCadJ/releases/tag/v0.24.6_gamma

It is a preliminary version, but it should fix the problem you have described in this Issue.
Let me know if ellipses can be now be selected easily (try very small as well as very large ones).
Cheers,

D.

Cipollino, 2016-03-30:
Hallo Dear DarwinNE, I downloaded FidoCADj v0.24.6 gamma and the issue of difficulty of selection object ellipse or circle is resolved now.
This issue is cleared.

Remain the issues on select an ellipse or rectangle and the CTRL key.
If I drawing an ellipse or an rectangle and hold the CTRL key, the figure obtained is circle or square.
If I select the up mentioned ellipse or rectangle, and I hold the CTRL key, the figure not remain an circle or square, but return in ellipse or rectangle.
Is possible resolve this second issue?

Many thanks for your patience.

DarwinNE, 2016-03-30:
Thanks, I will have a look at it.

Cheers,
D.

DarwinNE, 2016-04-04:
In the last commit, I should have treated the remaining part of the issue.
@Cipollino I am closing it, but feel free to manifest yourself if the problem is not solved: I will reopen it.
Cheers,
D.

closed 2016-04-04 by DarwinNE

Cipollino, 2016-04-06:
ALL OK now.