FidoCadJ/FidoCadJ #81: Little bug on drawing line and ADAPT view

Cipollino, 2016-03-30:
Dear DarwinNE, in Fidocadj v0.24.6 gamma (but the following strange comportment is present in all version of FidCADj) if I put the zoom view at 100%:

- I drawing an LINE;
- I double click on his, and in the propriety of the LINE, I set the dash style ................. (the third from the high in the list window);
- If the view is at 100% the LINE is OK, but if I hold the ADAPT tool (or maximize the zoom view to 300% ÷ 400% and plus) the LINE selected is again on continous tract.

The others LINE styles is visualized all OK, in all zooming settings, but the dash style ................ is not visualized OK from zoom settings of 300% and plus.

Is possible correct this wrong visualization of ............. dash style, on zoomed visualization?

Many thanks for your interest to my problem.

DarwinNE, 2016-03-30:
Thanks for the report, I think I see what you mean. I think it is just an issue in the calculation of the dashing parameters, I will see what I can do.

Cheers,
D.

DarwinNE, 2016-04-03:
Hi @Cipollino 
Version v0.24.6_delta should partially correct the bug about dashed lines:
https://github.com/DarwinNE/FidoCadJ/releases/tag/v0.24.6_delta

However, dashed lines are still not printed correctly. We will correct that in the next few days.

Cheers,
D.

Cipollino, 2016-04-03:
Dear DarwinNE, the v0.24.6 delta, in view mode, is OK now.
I'm waiting for your news.

Many thanks.

DarwinNE, 2016-04-03:
Ok, I think that with the last commit the Issue is fixed even when printing. I close this issue, if there is any related problems, we will open it again.

closed 2016-04-03 by DarwinNE

Cipollino, 2016-04-06:
ALL OK now.