FidoCadJ/FidoCadJ #82: "Save as" dialog window on Windows (8.1)

MigthyThor, 2016-03-30:
In Windows 8.1, with FidoCadJ 0.24.5 is not possible to past a path when you have to save a file in a **truly complicated location** ;-)

Finding the location in this way, using the tree, results a little bit annoying.

Can you consider changing the "Save as" dialog box in a better solution that allow a "paste" operation on the folder dorpdown ?

May be it's my bad and I didn't find the way to do it in the current dialog box.

Cheers!

DarwinNE, 2016-03-30:
Hi @MigthyThor 
  on Windows, FidoCadJ employs the Swing file dialogs.
A few years ago, we found that it was preferable to use native file dialogs on MacOSX. I think it may be the same for Windows. If you want to give them a try and you can compile FidoCadJ from sources, I can describe how to proceed to have them on Windows. If the result is better, we can employ them for the official versions.

Cheers,
D.

MigthyThor, 2016-04-01:
Hi @DarwinNE ,
sorry but I'm not capable of that at the moment. I don't compile complex software from 4-5 years.

My programming skills is so rusty now. May be in the future...who knows?

DarwinNE, 2016-04-01:
Ok. I will try to active that in FidoCadJ 0.24.6 delta I plan releasing in one or two days. Let me know if it works better.

Cheers,
D.

DarwinNE, 2016-04-03:
Hi, have a look at the release v0.24.6_delta:

https://github.com/DarwinNE/FidoCadJ/releases/tag/v0.24.6_delta

It should employ native file dialogs when operating on Windows.
Cheers,
D.

DarwinNE, 2016-04-06:
Some people reported that the older Swing dialogs are indeed preferable.

So version 0.24.6 will employ the following strategy

-   Windows: Swing dialogs
-   MacOSX: Native dialogs
-   Linux: Swing dialogs

MigthyThor, 2016-04-08:
Good!

DarwinNE, 2020-04-30:
Is this issue still relevant or it can be closed?