FidoCadJ/FidoCadJ #84: Print layers improvements

Cipollino, 2016-04-02:
Dear DarwinNE, I have reveled an incongruence from instrution manual in italian language and the print option in the FidoCADj v0.24.6 gamma.

In italian manual for v.0.24, in the cap. 2.3 I Layer, is letterally reported:
""Il nome ed il colore di ogni layer si può specificare attraverso il menu “Vista/Layer”. Dallo stesso menù si può decidere se disegnare o meno un certo layer a schermo, oppure in stampa.""

But in the CTRL+L menù, the important PRINT options, for the layer selectioned, is not present.

Also in the PRINT menù selection options, the print of "one layer only" is too poor, but is very important the possibility, from PRINT menù, of selection many other layer or ALL LAYERS.

This possibility is important for the following reasons:

FidoCADj is possible employed also in the simple technician in the mechanic drawing.

In the mechanic drawing is very important the control of layers, in the print and in the drawing phase, for  visualize or hide the different side of the work and differentiate the caracteristic of work phase (like: electric side from mechanic side, realized from various other external operators).

But this options is important also in the electronic PCB design.

Dear DarwinNE, is possible integrate, in the future version of FidoCADj the possibility, in the PRINT menù, of select :
- one layer only (like now);
- various layers (with flag in an square check box);
- all layers (with flag in many square check box);

this is for the perfect control of print in FidoCADj in the various phase of drawing.

Many thanks for your work.

DarwinNE, 2016-04-03:
Dear @Cipollino,
  thanks for the report.
The problem is that I only have the time to get the English manual up-to-date.
I should find someone willing to maintain the Italian version!
Thanks for the advices about printing multiple layers, it is a nice idea.

Cipollino, 2016-04-03:
Dear DarwinNE, for first many many thanks for your patience on receiving my proposals.
In my little opinion on italian manual of FidoCADj is the following.
Develop the english version only, for three big reasons:

1) your software is utilized from many people in THE WORLD (in the electronic field, but not only), and in the world the first language is ENGLISH (not the italian). Only in Italy the people learning and writing an second language is very few (also in the student people, and this is unacceptable in the 2016 - An good Architect of my knowledge not speak one word on English);

2) Learning and writing the English text is an personal improvements, much important for the searching in the internet world, important for electronics, but also for medicine, cure of strange diseases, etc., and push the people on this part means push for the personal knowledge and enlightenment, against a power that wishes to suppress the knowledge. 

3) Your time is LIMITED, and the cure of two manuals is an time taken away at improvements and debug of your very good work. 

Dear DarwinNE I'm italian, at school I have learning French language, but I write and learning with you in English (also with the help of Google Traduttore), for allow the knowledge and the exchange of ideas between me, you and the rest of the world.

I like the original language, but I recognize that most of the important news in the internet, as well as various software, are written in English and that's why I try to learn it better and better.

This is my humble opinion.

Many Thanks again for your patience.

DarwinNE, 2016-04-03:
Hi @Cipollino 
I got the same conclusion a few years ago. I am bilingual with Italian and French, but all the technical literature now is in English, so you must learn it if you work in a technical field.
Unfortunately, due to my limited time, now the Italian and French manuals are lagging behind, and I could only keep up to date the English manual.
Cheers,
D.