FidoCadJ/FidoCadJ #85: Missing text on landscape printing

Max2433BO, 2016-04-03:
Hi DarwinNE,

if you set "Horizontal page" in print menu box, all text in the page disappear.

Example:
If you print this

> [FIDOCAD]
FJC B 0.5
MC 105 50 0 0 830
MC 85 65 3 0 200
LI 55 70 105 70 0
LI 55 45 105 45 0
LI 105 45 105 50 0
LI 85 45 85 50 0
LI 85 65 85 70 0
LI 105 65 105 70 0
MC 55 45 2 0 010
MC 55 70 2 0 020
MC 135 50 0 0 750
LI 112 57 133 57 0
FCJ 0 0 3 2 1 0
LI 145 30 145 85 0
LI 135 65 135 85 0
LI 135 50 135 30 0
LI 155 30 155 85 0
TY 135 20 4 3 0 1 0 * L
TY 145 20 4 3 0 1 0 * N
MC 155 20 0 0 040
TY 130 10 4 3 0 0 0 * Dalla presa di rete
TY 130 95 4 3 0 0 0 * Al sistema audio
TY 25 35 4 3 0 0 0 * Pin 1 USB
TY 25 75 4 3 0 0 0 * Pin 4 USB
SA 85 45 0
SA 85 70 0
TY 80 75 4 3 0 1 0 * 1N4007

landscape printing is as follows

[image]

Instead, the vertical printing is normal

[image]

I encountered this problem on the following FidoCadJ versions:

> 0.2.24
0.2.25
0.2.26_gamma

Note for 0.2.26_gamma version: On the Print Preview window, the text always appears.

My test conditions are:

> O.S. version: Ubuntu 14.04 LTS 64 bit - Linux 3.13.0-83-generic
Java version: 1.7.0_95
OpenJDK Runtime Environment (IcedTea 2.6.4) (7u95-2.6.4-0ubuntu0.14.04.2)
OpenJDK 64-Bit Server VM (build 24.95-b01, mixed mode)
Printer: EPSON Stylus SX610FW
Printer driver version: CUPS STP01187.PPD 5.2.10-pre2

This is all!!!

Bye, Max

Max2433BO, 2016-04-03:
Update:

View the most recently version released today, replace above the** 0.24.6_gamma** version with **0.24.6_delta**.

Bya, Max

Max2433BO, 2016-04-03:
2th update:

Printing vertically this drawing (http://www.electroyou.it/forum/viewtopic.php?t=31581&p=641308#p641308 post [#302] @ mir) text does not appear, however.

Print test:
[image]

Bye, Max

Cipollino, 2016-04-03:
Dear DarwinNE & Max2433BO, I charge the entire design code in my FidoCADj v0.24.6.delta and I print it in orizontal or vertical layout page of my printer, and I not have problem in printing text.

My O.S. is WindowsXP and Epson Stylus Photo R265.

DarwinNE, 2016-04-03:
Dear @Max2433BO 
I tried to print the drawing and divert the result on a PDF file (it is very easy to do that on MacOSX); as @Cipollino, I obtain a correct result too.
Can you please try another font? Maybe "Courier New" is not installed in your system (the default font), the rendering on the screen is done with a substitution font and for an unknown reason this fails when printing.

Cheers,
D.

Max2433BO, 2016-04-06:
Hi DarwinNE,

I'm sorry but answer only now.

I don't think is a font problem, because the first example is printed vertically correctly (only in landscape printing missing text...).

Anyway I replaced the Courier New with Verdana (for example) and I get the same results of the described just above.

... I'm sorry.

Bye, Max

Note:
Identical result with 0.24.6_epsilon version.

Cipollino, 2016-04-06:
Dear DarwinNE & Max2433BO, I recharge again the entire design code in my FidoCADj v0.24.6.epsilon and I print it in orizontal or vertical layout page of my virtual printer doPDF and EPSON STYLUS PHOTO R265 = I not have problem in printing text.

My testing O.S. is WindowsXP & Windows8, but the result is the same = NONE PROBLEM

Very stranger Max2433BO.

Dear Max, is possible to you downloading and upgrade the Java virtual machine for avoid an eventual corrupted file ?

Bye for now.

Max2433BO, 2016-04-07:
@Cipollino 

I tried the latest version of Oracle jre available for Linux (1.8.0_77) and, unfortunately, the results are identical with last version of FidoCadj.

I don't know what else to try.

Bye Max.

DarwinNE, 2016-04-07:
@Max2433BO described in issue #90 that this appears to be an issue related to OpenJDK and not to FidoCadJ, so I close this issue.

closed 2016-04-07 by DarwinNE