FidoCadJ/FidoCadJ #88: Missing 1° caracter on Zoom field

Cipollino, 2016-04-04:
Dear DarwinNE, in the zoom field (in the left up side) the first character of indication of zoom is missing if the zoom is higher to 999% (the 1000% is visualized 000% - the 1300% is visualized 300%, etc.)

Dear DarwinNE is only an problem of the alignement to right of the text and, if is possible, enlarge the field of zoom indicator for avoid same problem with different screen resolution monitor.

Many thanks.

Cipollino, 2016-04-04:
Excuse me, I not indicated my version of FidoCADJ:   is  v0.24.6 delta.

DarwinNE, 2016-04-04:
Hi,
  I increased the size of the box by 20% It remains to see if that is OK with very high resolution monitors.
I will reopen the Issue if necessary.

D.

closed 2016-04-04 by DarwinNE

Cipollino, 2016-04-06:
All OK now.