FidoCadJ/FidoCadJ #90: Missing lines/dots with scrolling left/rigth

Max2433BO, 2016-04-06:
Hi DarwinNE,

I'm your nemesis... ;-)

I encountered this problem from version 0.24.4 to the 0.24.6_epsilon:

[image]

if you move the drawing to the left and right with the cursor to the bottom edge of the window, some parts are not drawn.
Instead, if you move the drawing in up or down with the cursor to the right edge in the window, there are no design problems.

This problem is also present in the text-only menu windows (example the menu window "open files"):

[image]

My test conditions are the same of the issue #85 .

Bye, Max

Cipollino, 2016-04-06:
Dear DarwinNE & Max2433BO, with FidoCADj v0.24.6 epsilon on my O.S. WindowsXP and video board Nvidia, I recharge the drawing and test:

1) up & down with mouse weel = NOT PROBLEM in viewing;
2) up & down with right-left key = NOT PROBLEM in viewing;
3) I put zoom to 2000% and ADAPT key = NOT PROBLEM in viewing;
4) I open dialog box for for research and open an file = NOT PROBLEM in viewing.

At this point I suggest an recharge of JAVA runtime machine (for eventually an stranger corruption of an file).

For the problem in visualizing drawing of ""cilindro a doppio effetto"", is an problem of video board????

The same result is with my other PC SAMSUNG with Windows8 and 13" monitor with HD resolution.

Much stranger Max2433BO.

Dear Max, is possible to you  download an old version of FidoCADj for replace the last version and test it  if the problem is visualized like now?

If the problem persist with the oldest version, the problem is in your video card.

Is possible this test?

DarwinNE, 2016-04-06:
Hi,
  thank @Max2433BO very much for the bug report and thank to @Cipollino for the test.
I currently think that it may be more a problem of the JRE you are currently employing because I have not written any code concerning the File dialog.
If it is possible, try to switch from OpenJRE to a version of it provided by Oracle to see if things are improving. In the past, it was not unheard that OpenJRE showed some bugs.
Or maybe try to see if the issue with the scroll appears also in other Java programs (some may employ the same File dialogs, which con Linux are the Swing versions).

Another possibility may be that FidoCadJ contains an error in some configuration procedures and that error determines the bug you are experiencing in your very specific case. However, as I said I am more inclined to think that it is a Java-related issue here, especially if the same behavior is shown with an older version of FidoCadJ as @Cipollino suggested (maybe try version 0.24.3 which was a pretty solid one, requiring only Java 1.5).

Kind regards,
D.

Max2433BO, 2016-04-06:
Hi,

I did the following tests:

I tried the 0.24.3 version, as suggested by **DarwinNE**, and the image problem is not present, but the problem on the text-only menu windows persists.

I tested another program with jvm (GeoGebra) and I have the same problem but only on the text-only menu windows.

I tried to download and reinstall the jvm, but without any improvement (As soon as I have a bit 'of time I'll try another version of jvm).

The problem of the video card, proposed by @**cipollino**, could be limited to programs that work with the jvm? (I haven' t no other program that present these problems)

Note: My video card is Radeon HD 5850 on monitor 1920x1080 W2486L LG.

Bye, Max

Max2433BO, 2016-04-07:
Updating:

I tried the latest version of Oracle jre available for Linux (1.8.0_77) and, unfortunately, the results are identical.

Bye, Max

Max2433BO, 2016-04-07:
Updating:

I tried to test the latest GeoGebra version (5.0.223.0) and there is no longer any scrolling problem.

Bye, Max

DarwinNE, 2016-04-07:
I tried FidoCadJ 0.24.6 epsilon on a Linux machine:

CentOS Linux release 7.2.1511 (Core)
openjdk version "1.8.0_71"
OpenJDK Runtime Environment (build 1.8.0_71-b15)
OpenJDK 64-Bit Server VM (build 25.71-b15, mixed mode)

I could use it by connecting my means of ssh and by using a X server on my MacOSX laptop.
I could not reproduce the bug: I could load drawings, modify them (albeit a little slowly, due probably to the network connection) and scroll them without any issue.

Max2433BO, 2016-04-07:
Hi,
I may have solved the mystery !!

I deleted all references to OpenJDK present on my computer (before I had just reinstalled the sotfware), so I started FidoCadJ 0.24.6 epsilon with Oracle version of Java (the only one on the PC) ...

... And it worked perfectly !!!

So I reinstalled OpenJDK and the problem has come back (also using the Oracle version of java): at this point could be a problem of the OpenJDK version for Ubuntu, I will try another version and try again! 

Note: With only one version of Oracle java installed also the problems presented in issue #85 you solve!

Bye, Max

DarwinNE, 2016-04-07:
Great news, thanks so much @Max2433BO 
I can close this Issue here.

closed 2016-04-07 by DarwinNE