FidoCadJ/FidoCadJ #92: Linux distribution specialist

DarwinNE, 2016-04-10:
Hi to all,
  I am seeking for someone willing to rationalize the way FidoCadJ is distributed on Linux.
Here is what this person should do:

- choose the best strategy for distributing FidoCadJ on Linux
- answer to Linux-related questions
- prepare the distribution files each time a stable release of FidoCadJ comes out
- promote the software on Linux communities, propose and follow the integration on major Linux distribution if possible

Is there anyone willing to do that?

simon-zz, 2016-04-12:
Hi,

Regarding the installer, I think we could start having a direct link to the latest release, and maybe for all OS releases. This makes easier to download from a script, without changing the script every time a new release will be released.

Then, having an icon available, the script could determine the desktop environment used and creating a menu launcher (using the icon). This maybe will be not so tricky as downloading FidoCadJ and just putting in a system folder.

In my opinion, I would leave docs available to be downloaded manually from FidoCadJ website, adding a [ Doc ] link there beside to the Download section.

Regards,
Simon

Stemby, 2016-04-13:
Here is the direct link to the latest release:

https://github.com/DarwinNE/FidoCadJ/releases/latest

Ciao!

simon-zz, 2016-04-13:
Well, I suggest a direct link which would be useful to download the jar file through wget ... : )

Simon

Stemby, 2016-04-13:
$ wget `curl -s https://api.github.com/repos/DarwinNE/FidoCadJ/releases/latest | jq --raw-output '.assets[0] | .browser_download_url'`

simon-zz, 2016-04-13:
Thank you, but I don't have **curl** neither **jq** installed so I cannot test the command you posted.

I would like to write an installer with minimum requirements: java and wget. Nothing more, nothing less. 
I will suppose that basic GNU utils and find-utils will be installed on the distro used. 
No need to use 'which' to locate packages, for example.

So, it's better to have a file (or symbolic link to but I have to test) always  pointing to last version.

Regards

Stemby, 2016-04-13:
$ wget `wget -qO- https://api.github.com/repos/DarwinNE/FidoCadJ/releases/latest | grep browser_download_url | head -n 1 | cut -d '"' -f 4`

simon-zz, 2016-04-13:
That's better. Thumb up.

DarwinNE, 2016-04-13:
A script is a nice idea, but I suspect that the vast majority of Linux users tend to install things from their distribution. What about proposing to include FidoCadJ in the major Linux distributions?

Cheers,

D.

Max2433BO, 2016-04-17:
Hi,

As an "very" end user, I agree with the @DarwinNE idea's tied for the Linux distribution installation in use (.deb, .rpm, ...) or the creation of a stand-alone format.

I realized how example that has not been tested for long, but it seems to work, contains everything you need to run FidoCadJ:

Jre1.8.0_77
fidocadj_0.24.6.jar
fidocadj_0.24.6.sh

all in a subdirectory called FidoCadj

The fidocadj...sh is structured as:

> #!/bin/bash
export PATH=.:../Fidocadj/jre1.8.0_77/bin
export CLASSPATH=.:../Fidocadj/jre1.8.0_77/lib:../Fidocadj
../Fidocadj/jre1.8.0_77/bin/java -jar ../Fidocadj/fidocadj_0.24.6.jar

and is selected as executable.

Note: The .sh structure is not "my bag of flour" but is derived from the following article:

[http://www.electroyou.it/alev/wiki/fidocadj-portabile-un-cae-al-guinzaglio](url)

Bye, Max

DarwinNE, 2016-04-28:
Hi to all,
  any news?
Can we summarize things? What is thus the best distribution strategy to be applied for Linux? Who wants to take the action?

Cheers,
D.

Stemby, 2016-04-28:
In my opinion the best strategy is having FidoCadJ included in official repositories, at least Debian and Fedora.

I'd like to be the Debian maintainer, but currently I can't. Maybe in some months...

DarwinNE, 2016-06-03:
Hi @Stemby,
  do you think we can do something soon?

Cheers,

D.

P.S. by change, do you know any FreeBSD user for Issue #100 ?

Stemby, 2016-06-08:
Hi @DarwinNE,
sorry, currently I can't do anything soon, and I don't know any FreeBSD user.

Stemby, 2016-06-16:
Hi,
[this](http://snapcraft.io/) package format might be a very good choice.

DarwinNE, 2016-06-16:
Hi @Stemby 
  that sounds very nice!
@Max2433BO and @simon-zz what do you think about that?

Kind regards,
D.

simon-zz, 2016-06-17:
Hi ! I didn't know it !

Honestly, I never made a package for any distribution just because I never needed it..
On the next weeks (when I will have some time) I could try to make some test for Debian (the one I am actually using). 

It should not be so hard because it's plenty of 'deb packaging how to' out there, and also because the only one dependency of fidocadj is java, which maybe have its own dependencies.
I will try snapcraft too.

PS: fidocadj will need a repository dedicated to fidocadj and each user will have to add the repo line to the package manager configuration file. 

Cheers,
S

Stemby, 2016-06-17:
Hi @simon-zz,
instead of a "private" repository (Snaps should replace unofficial repositories) what do you think about maintaining an official Debian package?

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=708312

Ciao!

DarwinNE, 2016-06-19:
@simon-zz @Stemby if you succeed in putting together an official Debian package, I will officially be in total awe of you 🎱 👍 

D.

DarwinNE, 2016-06-26:
Hi @simon-zz @Stemby 
do you think it would be possible to try to do that?

Cheers,
D.

simon-zz, 2016-06-29:
Hi Davide,

I will try after the next week. 
I can't do it before (I have to solve other problems urgently :( ).

Cheers,
Simon

DarwinNE, 2016-06-30:
Great 👍

simon-zz, 2016-08-07:
Hi @DarwinNE and @Stemby,

Pleae check out my repo **fidocadj-debian-package-builder**. 
I have added the files to make a debian package. 

This is a first basic version.. :dancers: 

Please review the description of the package insinde the DEBIAN/control file.
In that file, the line next to the **Descripton:** marker is a short description. The following should be a long description. @DarwinNE please review it and suggest me what to write as description there.

Test it, of course.
Cheers,
Simon

DarwinNE, 2016-08-12:
Hi @simon-zz and @Stemby 
I can not try it as I do not have a computer running a Debian distribution, but from what I saw it seems to me that you did a great work as usual. I had a look at the description and corrected a few things, I tried to open a pull request, let me know if it works.
Maybe we could add a link to the webpage of the project: [http://darwinne.github.io/FidoCadJ/](http://darwinne.github.io/FidoCadJ/)

Cheers,

Davide

simon-zz, 2016-08-12:
Hi Davide, thank you, I just try to do my best, which is not debian packaging of course :( ..

Before adding a link, I would like to improve the package for adding menu icon access from the desktop environment. The problem is I don't know them all. I used Gnome a few years ago, but since then I am happy with Xfce ... 

I would appreciate if someone would suggest me where the *.desktop files are located into the system using a KDE DE etc.. As I expect nobody will suggest me something (worst case), I will run a KDE live distro and look how it works ... 

Anyway, I will take a look at it next days.
Regards,
Simon

Stemby, 2016-08-12:
@simon-zz: both Gnome and KDE (and XFCE, and [...]) should follow the freedesktop.org standards.

https://developer.gnome.org/integration-guide/stable/desktop-files.html.en

Best regards,
Carlo

nivit, 2016-08-13:
@simon-zz 

`sudo desktop-file-install fidocadj.desktop`

if you have installed the package `desktop-file-utils` on your system.

simon-zz, 2016-08-14:
Thank you guys ! I will try to automate all next days. 

Regards,
Simon

simon-zz, 2016-08-15:
Hello guys,

The files have been updated to create (and removing after installation) for desktop menu launcher.
Please, test it.

Thank you,
Simon

ekardnam, 2020-04-30:
Althought this issue is old, it's still open so...

There is a valid PKGBUILD in the AUR at https://aur.archlinux.org/packages/fidocadj/ .
A PKGBUILD is the kind of file Arch Linux uses to specify how a software should be packaged into a valid package, it's basically a bash script.

The current PKGBUILD downloads the latest release from sourceforge and installs it (more info on Arch Linux packages and the PKGBUILD format at https://wiki.archlinux.org/index.php/Creating_packages). Actually the PKGBUILD can also build the from sources and in case of a package in the AUR it actually should ("The packages on the AUR are merely "build scripts", i.e. recipes to build binaries for pacman." from https://wiki.archlinux.org/index.php/Arch_User_Repository#FAQ).

Hope this information is useful. We can try contacting the author of the PKGBUILD or I can make a new one for new releases!

DarwinNE, 2020-04-30:
Hi @ekardnam 
Thank you very much, it will be very useful.
I don't have a Linux machine currently available for preparing the distribution, I will need some help for that.

By the way, I had a pleasant surprise with the jpackage tool in Java 14. It can not cross-produce packages, but it seems to work quite well on MacOS. I think it can produce .rpm packages for Linux, maybe it is worth a try.

Cheers,
D.

ekardnam, 2020-04-30:
Hello @DarwinNE 
For as long as it reguards Arch Linux the AUR I think is as much as you can get right now. Usually most used packages in the AUR get into the community repositories. We could make a non official repository but it would really make sense as one would have to add this repository.
Though the AUR is much used by Arch users and for people who use yaourt or yay or other pacman wrapppers they can install fidocadj with a single command like `yaourt -S fidocadj`. And many people use this kind of wrappers (yaourt is default on Manjaro, yay is one of the most voted packages in AUR).

So I guess that for Arch based distros that is as far as we can get right now.

I don't have much experience with Rad Hat. We could try docker to test whether the produced rpm works

ekardnam, 2020-05-01:
Hello @DarwinNE 

It already in the AUR (https://aur.archlinux.org/packages/fidocadj/) packaged by a guy called danitool. He/She packged the last release in 2017. I can try contacting him if you want.

I can package new releases if necessary. If danitool does not reply when a new release is issued we can mark that package as being not updated and create a new one

DarwinNE, 2020-05-01:
Hi @ekardnam 
   ok for me. Let's wait for when the 0.24.8 will be ready.

D.

ekardnam, 2020-05-01:
Hi @DarwinNE 
I have written to danitool hinting him to this issue.

ekardnam

DarwinNE, 2020-06-13:
Hello,
  I just published the stable version 0.24.8: https://github.com/DarwinNE/FidoCadJ/releases/tag/v0.24.8

ekardnam, 2020-07-04:
Sorry DarwinNE, I have quite a lot of university exams. Gonna take a look at this in a few weeks if nobody else does :)

DarwinNE, 2020-07-04:
Hi @ekardnam 
no problem! If you can do something, that would be wonderful!

Cheers,
D.

DarwinNE, 2020-12-04:
Hi @ekardnam 
Could you do something, by chance? :-)

Version 0.24.8 seems to be relatively stable and appreciated, it may be a nice include in a Linux distro.

Cheers,
D.

ekardnam, 2020-12-09:
Hello @DarwinNE I'm gonna look at it next week is that okay?

DarwinNE, 2020-12-09:
Hi @ekardnam that will be great.
Thank you very much!

ekardnam, 2020-12-19:
I'm looking at what the best options are for distributing on all distros.
I've seen mentioned snap which is cross distribution. An other alternative would be AppImage.
I'm doing and publishing on the AUR the package for arch linux (unless someone already did that which could be possible), as for other distributions like debian I'm not sure whether a user would rather download the .deb and install it when they can just download the jar file, so I was thinking if having a deb could be worth it. I think it would be worth it only if the file was in the official repositories so it could be installed from terminal without extra steps needed.

Snap would allow us to publish it on snap repositories and have it install with `snap install fidocadj` if the user has snap installed. AppImage would just require the user to download the AppImage file and double click it to run it, with all the dependencies self contained in the appimage file.

Im gonna look more into what is needed to get a package into the official debian repositories and same for red hat as well as put it into the AUR (which is the gateway to get into the Arch Linux repositories) tomorrow (university got me busy all week)

ekardnam

DarwinNE, 2020-12-19:
Thank you, @ekardnam!

JoopN, 2022-06-13:
Probably won't work with Raspberry Pi systems, I own a RPI400. For some reason no word how to get Java programs going and how to get it into your menu program. I did ask on the Pi forum and got back that they don't like Java and favor C+ and Python. Wrong answer if you ask me, both are not portable between different OS-systems. So I had to find out myself as  usual. Linux sucks (sorry) when it comes to file management. Parts of a program are all over the system with as result you can't find a thing. So this is my approach:
Download the Java program as zip or jar file if available. Go to the file manager and go the home/pi directory. Make a new directory "Apps" or "Applications" or something which you like. Open this new directory. In this case make a new directory "FidoCadJ". As FidoCadJ comes as a jar-file, copy the file to this directory. Now the magic. Open Mousepad. Copy next lines into it and save it as fidocadj.sh. Omit the lines with "# start" and "# end". This setup works for all Java programs up to now, of course the line java... is then somewhat different.

# start 
#!/bin/sh

#go to script's directory
progdir=${0%/*} 
cd $progdir

#launch fidocadj
java -Duser.home=$progdir -jar fidocadj-0.24.8.jar 2>fidocadj-bugs.txt 
# end

As I'm translating the manual I need sometimes FidoCadJ not in my native, the I use this script;

# start
#!/bin/sh

#go to script's directory
progdir=${0%/*} 
cd $progdir

#launch fidocadj
java -Duser.home=$progdir -jar fidocadj-0.24.8.jar -l en 2>fidocadj-bugs.txt

# end

This setup gives you lots of freedom and more important I can find it on my system.