FidoCadJ/FidoCadJ #96: Small components libs

simon-zz, 2016-04-12:
Hi Davide,

As we organized, I open this issue as a request to include my library of 'small' electronic symbols ( which can be found here:  https://github.com/simon-zz/simon-zz-fidocadj-libs ) in a future FidoCadJ release.

Regards,

#Symbols 
#Website

DarwinNE, 2016-04-12:
Hi @simon-zz 
thanks very much, we see if we can put it here for the moment:
http://darwinne.github.io/FidoCadJ/libs.html

Cheers,
D.

simon-zz, 2016-04-12:
I can't figure out how to put labels... Sorry :(

DarwinNE, 2016-04-12:
BTW, can you provide a short description of the library?
Which license do you choose? CC BY-NC 4.0 is ok for you?

Cheers,
D.

P.S. I just added the labels, no problems there!

simon-zz, 2016-04-12:
This is a library I started to create because I wanted small and a minimum number of symbols for my schematics. 

I reduced the size of passive components, OpAmps and Voltage Sources.
Transistors, electrical ground and GND reference symbol are taken from EYLibraries : ) , except a very very small change in BJTs and Darlingtons.

 CC BY-NC 4.0 Licence is OK. 

Regards
Simon

MigthyThor, 2016-04-13:
So, we can upload the library on the web site and add it to our page: 
http://darwinne.github.io/FidoCadJ/libs.html

All I need are the correct informations to append a new row to the table: file name and link, weight, version, description in english and in italian.

When you two are ready, write me all these parameters.

Have a nice day!

DarwinNE, 2016-05-12:
Hi @simon-zz and @MigthyThor 
It would be very interesting to have the library in the website. Were you able to do something about it?

Kind regards,
D.

simon-zz, 2016-05-13:
Hi Davide,

Thank you for your interest. Sorry, I have been very busy during these days and I cannot think on it.
I'll try to do this weekend what MigthyThor suggested.

Regards,
s.

DarwinNE, 2016-05-16:
Hi @simon-zz 
Thank you very much! 
Let us know when you can do that! 

Cheers,
D.

simon-zz, 2016-05-19:
OK guys so here I am.. +1: 

I would write, into the describtion: 

"Proposed by simon-zz on Git Hub. This is a small and basic library of electrical and electronic symbols. The symbols has been designed to be reduced in size respect to the other libraries, allowing the user to design small size schematics."

The actual size is 15983 bytes and the version is ... let's write 1.0, since it would be a first public release.

The library, as stated in the first post of this discussion, can be found in my repos at the link in the same post. I updated it an hour ago: I added a Smith Chart (as complete as possible :) ) and a Smith Chart plane (without reflection and transmission coefficients, return loss etc. scale and wavelength references on exterior circumfereces), perhaps more convenient to use in certain situations.

Please let me know if you need more infos.

Regards and thank you again for your interest.
Simon

DarwinNE, 2016-05-19:
Hi @simon-zz 

Seems very good, thanks!

MigthyThor, 2016-05-20:
So, are we ready to upgrade the libs page on website? (not before the next week, sorry)

simon-zz, 2016-05-21:
Yes, I think we can release thi 1st vesrion.
What about future updates ?
Cheers
Simon

DarwinNE, 2016-05-23:
Hi @simon-zz and @MigthyThor 
Thanks for the great work!

I think we can release the first version on the website. When that will be done, we will close this Issue.
For the moment, for updates, probably it will be simpler to  reopen this issue, update the website and close again when all is done.

If updates are too frequent (for this, or for other libs), we may think about a specific GitHub project, or maybe a database system of some kind.

Cheers,

D.

DarwinNE, 2016-06-03:
Hi @MigthyThor 
do you think you can do the update soon for the website, if you have some time?

Kind regards,

D.

MigthyThor, 2016-07-22:
Done! Cheers

DarwinNE, 2016-07-24:
Hi @MigthyThor and @simon-zz 
I merged the pull request by @MigthyThor.

Terrific work, as usual!

Cheers,
D.

DarwinNE, 2016-07-24:
I close this Issue, we can reopen it if a new version of the library becomes available.

closed 2016-07-24 by DarwinNE

simon-zz, 2016-07-24:
Great ! Thank you guys !!!