FidoCadJ/FidoCadJ #98: Print margins

Luigi-C, 2016-04-23:
Ho installato la versione 0.24.6 di FidoCadJ.
Rispetto alla precedente la funzione di stampa è migliorata, in particolare l'adattamento alle dimensioni della pagina.
Ho riscontrato però un problema con la gestione dei margini di stampa.
La stampa finale presenta dei margini (corrispondenti ad un'impostazione predefinita di 2,55 cm), sia in senso verticale che in senso orizzontale, piuttosto cospicui, che lasciano inutilizzato del prezioso spazio sul foglio di carta.
Ho provato a ridurre i margini, ad esempio ad 1 cm, ma il circuito viene tagliato (vedi zona rosa dell'anteprima in "Opzioni di stampa") e non viene quindi stampato completamente.
Ho notato che cambiando stampante, ad esempio passando da una LaserJet ad una stampante virtuale tipo PDF Creator, i margini restando invariati, quindi immagino non dipendano dallo specifico driver di stampa.
Cosa è possibile fare per utilizzare pienamente il foglio A4?

DarwinNE, 2016-05-12:
The default print margins are calculated by taking the maximum printable area declared by the printing drivers handled by Java. In the print preview, FidoCadJ shows in red the region lying outside the declared printable area whereas the margins selected are shown by means of a dashed line. Here is what I get in my case (MacOSX 10.11.2 with a french locale):

[image]

It seems that in your case the declared printable area is abnormally small. Maybe a driver issue? Or a Java configuration problem? What is your operating system?

Luigi-C, 2016-05-12:
Hi Darwin, I tried to set another printer as the default one (PDF Creator virtual printer), but nothing has changed, the printable area remains the same:

[image]

I don't know Java in details and I don't know how to configure it to trim printing margins in FidoCadJ.
My OS is Windows 7 64-bit edition.

DarwinNE, 2016-05-13:
Hi @Luigi-C,
  it seems that in your case the margins are set to 1 inch (hence 2.55 cm) by the fact that the printable area is declared in this way by your Java system.
For the moment, I do not see any obvious error on the FidoCadJ side, but I will continue to investigate.
Cheers,
D.

closed 2018-02-18 by DarwinNE