FidoCadJ/FidoCadJ #99: How to trace on to an image

DarwinNE, 2016-05-10:
An old request which was open on Sourceforge was about the ability to select an image as a background to trace on it. It seems to be useful for a lot of people and I received a lot of requests about it.
A quick workaround is available here: https://github.com/DarwinNE/FidoCadJ/wiki/How-to-trace-images-with-FidoCadJ

However, implementing that directly on FidoCadJ might be discussed here.

Cipollino, 2016-05-13:
Hallo dear DarwinNE, if you is interested to my personal opinion, this is an very important implementation on FidoCADj for followings reasons:

1) Interesting is import any image (JPG - PNG - GIF) like background on FidoCADj table - For drawing in his surface of line or other geometrical figure;

2) interesting is also draw an electrical circuit directly from the electrical scheme, from an image in background (but in this case is provident the TRANSPARENCY function of FidoCADj,  for pointing exactly in the interceptions of components).

Many thanks for your interest to my proposal.

DarwinNE, 2016-05-13:
Hi @Cipollino 
My idea would be to avoid to import an image. In fact, it would be cumbersome to store a bitmap image into the text format which is employed by FidoCadJ.
My question is whether this would be ok or not. I would not like to change the main philosophy of FidoCadJ which is, among other things, to provide small, compact, text files.
Apart from that, it should be technically feasible to show an image on the background, but it will not be stored in the drawing. On the other hand, this behavior may confuse some users.

Cheers,
D.

Cipollino, 2016-05-13:
Yes DarwinNE, the image is not imported in the FidoCADj (for the light code interchange), but is possible draw and place an line, or an  other geometrical figure, on the background visualized on FidoCADj desktop.
For best explane the utility:
I working in an Police Department for reliefs of automobiles accident. 
Many times I placed on an background of the zone of accident and I put line of direction of vehicles, people and vehicles on road.
In this case is possible utilize FidoCADj again for this purpose, but the employs is limitless (imagine: architecture of internal and many others utilize of overlap).
In this case, the background is for referent pointing only .

In this case is good the opportunity of:

- resizing and drag the image background (for rearrange the image to an previous file saved in FidoCADj code light);
- exporting in final image visualized on the screen in JPG or PDF WITH BACKGROUND (or WITHOUT BACKGROUND) ALSO, but not save the background in the file code of FidoCADj.

This possibility is good also for draw the scheme of circuit, from an background content electrical scheme.

Also in this case is important the export of entire drawing  (WITH or WITHOUT the background and Fidocad points, lines and symbols) in JPG or PDF high resolution, but saving this scheme of circuit in FidoCADj file code only.

Excuse me DarwinNE for my exposition, if I not expose clear the concept. 
Naturally: this project is very challenging for anyone.

Is an idea only.

Many thanks for your patience.

erichVK5, 2017-02-20:
Once the support for a background image is working, I would like to test this in combination with the new module allowing export to gEDA PCB/pcb-rnd as a means of quickly and easily duplicating/reverse engineering existing PCBs.

DarwinNE, 2017-02-26:
Hi @erichVK5 
that possibility was intended exactly for that. I have been in vacation for a while, but I am planning to work on it in the next weeks so that a little bit of testing will be possible.
D.

closed 2017-04-18 by DarwinNE