FidoCadJ/FidoCadJ #100: FidoCadJ and FreeBSD

DarwinNE, 2016-05-24:
It seems that the FreeBSD package build server contains a FidoCadJ port which does not work.
Among other things, it still points towards Sourceforge, it would be better to have:
- an update version of the port pointing to a 0.24.6 version
- a link towards GitHub.

Here is a page describing the issue:
https://www.mail-archive.com/freebsd-pkg-fallout@freebsd.org/msg284993.html

nivit, 2016-06-04:
Hi @DarwinNE 

that error happens because you ship a distribution filename (fidocadj.jar) without a version number, so the checksum of old files fails. The FreeBSD ports is updated: [http://www.freshports.org/cad/fidocadj/](http://www.freshports.org/cad/fidocadj/)

DarwinNE, 2016-06-04:
Thanks!

So that problem should not appear with the GitHub release system, which include the version number in the file path?

Kind regards,

D.

nivit, 2016-06-04:
Actually, we would need a filename with the version such as fidocadj-0.24.6.jar.

DarwinNE, 2016-06-04:
I see. So the version number is still not enough:

https://github.com/DarwinNE/FidoCadJ/releases/download/v0.24.6/fidocadj.jar

We will think about adding the version (as for the Windows installer), but in the next version...

Cheers,
D.

DanteCpp, 2016-08-30:
I agree with @nivit, 
the version should be also in the package name. 
Now I am updating the developers' tools, so I could automate the naming of the package.

Cheers,
Dante.

DarwinNE, 2016-08-30:
Hi @DanteCpp and @nivit 
The real reason which is behind my reticence is not to rename the file: we will do that for the next stable release.

The real issue is I would like to simplify as much as possible the release of each new stable version, which is a sort of a PITA. For example, updating the website is not something I really like to do (we are lucky that @MigthyThor helps a lot for that). When we used SourceForge, for example, I appreciated that the last version was always at the same place with the same name as the links remain the same. Currently, when a new release is out, the links in the website have already to be changed but @MigthyThor made sort that this is a relatively easy work. However, after all I noticed that having the release number in the filename helps a lot.

Any suggestion about simplifying things for the release is much welcomed.

Cheers,
D.

DarwinNE, 2020-04-30:
FidoCadJ 0.24.8 will contain version number in the package filenames.

DarwinNE, 2020-05-08:
The FreeBSD package is not maintained here. The change in the naming convention should help the maintainers. I therefore close this feature.

closed 2020-05-08 by DarwinNE