FidoCadJ/FidoCadJ #111: The arrow length and half widths may be floating point

DarwinNE, 2016-08-29:
Hi to all,
In some cases, a fine control over the length and width of arrows would be beneficial.
Currently, those values can only be integer, that is a pity!

Version 0.24.7 should be able to parse double precision values for that (yet without letting the user only employ integer values). A complete support should be available for the successive version.

DanteCpp, 2016-08-31:
Hi @DarwinNE,
in order to draw double precision arrows, the [GraphicsInterface.java](https://github.com/DarwinNE/FidoCadJ/blob/master/src/net/sourceforge/fidocadj/graphic/PolygonInterface.java) interface should be updated. Consequently also its implementations must be updated. 

About the swing implementation a possible solution is [here](http://stackoverflow.com/questions/15620590/polygons-with-double-coordinates).

Cheers,
Dante.

DarwinNE, 2016-08-31:
Hi @DanteCpp,
  yes, I agree with you.
What I would like to do is however to integrate progressively the change in format. For example, this is not parsed in version 0.24.6:

`LI 30 40 110 40 0
FCJ 3 0 3 2.6 0 0`

So, here is the roadmap:
1. Version 0.24.7 will parse that code showing an acceptable result (even rounding up the size to 3 instead of 2.6).
2. Version 0.24.8 will show the correct result and will let the user enter a non-integer value.

Apart the obvious GraphicsInterface.java interface, all classes handling export towards a vector file format have to be updated.

Cheers,
D.

DarwinNE, 2016-09-01:
Hi
I implemented the first part of that issue, so now a non integer size is still recognized and the size is rounded up.

BTW, I had to change the three classes where the arrows can be present (PrimitiveBezier.java, PrimitiveComplexCurve.java and PrimitiveLine.java), so this suggested me that a little bit of refactoring could be beneficial to obtain a more compact code. I will probably work towards that in the next few days.

Cheers,
D.

DarwinNE, 2017-04-18:
I am assigning the remaining work to the FidoCadJ 0.24.8 milestone.

DarwinNE, 2018-08-05:
I think we can close this issue now.

closed 2018-08-05 by DarwinNE