FidoCadJ/FidoCadJ #117: Timeout for message window

Max2433BO, 2016-09-09:
Hi,

This isn't a problem but a feature request for new release of FidoCadj.

Synthetically the request is as follows:

1) I do a drawing
2) I save it and I export it (for example as EPS format)
3) I open the same file to correct it, then I save it
4) I export it with the same name (as in step 2)
5) Appears a window that ask if I want to overwrite (I click OK)
6) then a window appears that indicates to me that the file was saved...

... **but this last window will remain until you press OK**

Considering that it represents only a warning, not an alarm message, it could not do so after a few seconds closes without having to click on ok?

Bye, Max

Window in question in the Italian version
[image]

DarwinNE, 2016-09-09:
Hi @Max2433BO 
I agree, this dialog box is definitely quite annoying!

DarwinNE, 2016-10-01:
Hi @Max2433BO 
I think a notify in the coordinate region would do the trick.
It may be erased by the coordinates if the mouse is moved. Much less disruptive than the modal dialog you shown.

[image]

Cheers,
D.

Max2433BO, 2016-10-02:
Hi, @DarwinNE

It might be an idea, but do not be too out of sight with respect to the "export" dialog box?

After clicking on "OK" closes the window and not seeing anything special happen (the writing is, as mentioned, out of sight) I wonder: The file will be exported?

... But maybe I worry too much!

Bye, Max

DarwinNE, 2016-10-02:
Hi @Max2433BO 
  I do not like very much having a timeout in a modal dialog frame.
Maybe I can change of the message colour (green?) for a while.

Cheers,
D.

Max2433BO, 2016-10-02:
> _Maybe I can change of the message colour (green?) for a while._

Why not... ;)

Bye, Max

DarwinNE, 2016-10-02:
Hi,
  here is what I obtained:

[image]

That should be visible enough, in my opinion.
The colours became normal when the mouse is moved and the coordinates shown.

Cheers,

D.

Max2433BO, 2016-10-02:
Perfect!!! You're a big, @DarwinNE  :)

Bye, Max

DarwinNE, 2016-10-02:
Thanks @Max2433BO 

I renamed this issue for future reference, I am closing it now.

Cheers and thanks,
D.

closed 2016-10-02 by DarwinNE