FidoCadJ/FidoCadJ #120: Often I can't save new components in the libraries

Max2433BO, 2016-10-02:
HI, @DarwinNE 

I shall quote a problem reported by a mutual friend from electroyou.it site:

> when you create a new component for FidoCadJ (_I suppose the 0.24.6 version, but I'm not sure_) libraries and you save it, often does not accept the new component because the key number is already assigned.
In order to save the new component we must make a series of attempts by assigning a random number.
... And if you are unlucky, it happens sometimes, you have to try and try again several times. :(

This is ... I haven't tried it, and I will propose as is.

Bye, Max

DarwinNE, 2016-10-02:
Hi @Max2433BO 
If I understand correctly the problem, a solution may be to put the background of the key field in red if the key can not be accepted (because incomplete or replicate in the library).

Cheers,
D.

DarwinNE, 2016-10-02:
Here is my solution. An invalid key is shown in this way:

[image]

If the "ok" button is clicked anyway, the usual error message appears.
Now the program checks if the code is available or not before choosing one more or less randomly.

Max2433BO, 2016-10-10:
Hi, @DarwinNE ,

 it's a good idea for me ... now we have to hear what think our mutual friend ;)

Bye, Max

DarwinNE, 2017-01-14:
Ok, I close this issue now.

closed 2017-01-14 by DarwinNE