FidoCadJ/FidoCadJ #126: Proposal: To store the usage parameters program

Max2433BO, 2017-01-04:
Hi, @DarwinNE 

I send a request from a Electroyou forum user:

> Is it possible that fidocadj remember the parameters with which it was closed?

Specifically: location and size (expanded or reduced) of the window, setting zoom, grid, grid block, ...

DarwinNE, 2017-01-06:
Sounds reasonable!

DarwinNE, 2017-01-14:
FidoCadJ now saves:
- the size and the position of the window
- the visibility of the Library tree (see #127)
- the snap to grid state
- the visibility of the grid

I do not think that the current zoom must be saved too, since every time a new file is opened the zoom is recalculated and if one starts a new drawing it is better to start with a reasonable zoom setting (400%).

closed 2017-01-14 by DarwinNE