FidoCadJ/FidoCadJ #130: Proposal: Zoom with mouse wheel

Max2433BO, 2017-01-04:
Hi, @DarwinNE 

I hope you do not hate me since I entered 5 issue you in one day ... ;)

... hopefully it is the last (for now) you back a request: do you think it is possible to realize the zoom control with the mouse wheel, for example, similar to that available on Google Map?

Bye, Max

DarwinNE, 2017-01-06:
Cntrl+wheel is not working?

It is CMD+Wheel on MacOSX, it works but it is maybe a little too sensitive.

gammaci, 2017-04-25:
MAC – Zoom with mouse wheel malfunction REPORT

See (http://www.electroyou.it/forum/viewtopic.php?f=21&t=31581&start=430#p708492)

A minor malfunction has been observed concerning the Zoom function when using the mouse wheel. Please follow the steps sketched below to reproduce the problem:

1) Copy the circuit code below in your “Fidocad Code” panel and hit OK
2) Click “Adapt” icon, you should have two OP Amps feeding a mixer filling entirely your drawing area, now Zoom % = 1050
3) Click “Zoom” icon, the mouse pointer changes like a little pointing hand
4) The Zoom center point is exactly on the index fingertip, click on the center of the cross inside the mixer symbol, now Zoom % = 1575
5) Hold down CMD key and play with mouse wheel, e.g. Zoom In to 2000% then Zoom Out down to 395% - The cross inside the mixer stays always centered on the index fingertip.
6) Now Zoom Out from 395% to 263% - BANG! – The drawing moves unexpectedly to the left escaping from the Zoom center point. Any subsequent attempt to regain control of the previous perspective with ‘Zoom In’ fails. To regain the drawing one has to click “Adapt” and restart his/her work.

Not tested, but seems reasonable, the same behavior could be observed working on upper right and lower left/right corners of drawing area. 

`[FIDOCAD]
MC 85 75 0 0 ey_libraries.opaopa1
MC 110 100 3 0 ey_libraries.opaopa1
MC 110 75 0 0 ey_libraries.sigblk5`

DarwinNE, 2017-04-25:
Hi @gammaci 
Thanks for the procedure, I could reproduce the bug.

Cheers,
D.