FidoCadJ/FidoCadJ #131: Fidocadj 0.24.7. alpha: "Holes" in display tree libraries

Max2433BO, 2017-01-21:
Hi, @DarwinNE 

In 0.24.7.alpha version of Fidocadj we found a tree problem viewing of libraries: 

if you open the list of "Componenti discreti - Libreria standard" (Italian version) you will notice a "hole" on top of "Resistenza" and above "Resistenza variabile", but if you look inside the "Libreria standard - Componenti discreti", actually there are two types of "Resistenza" and "Resistenza variabile" (Note that the same open list with 0.24.6 FidoCadJ version does not have the same problem).

**Image 0.24.7 alpha version**
[image]

**Image 0.24.6 version**
[image]

Also this defect is present only in that section of the "Libreria standard" and in no other embedded library in the .jar file of the Fidocadj 0.24.7 alpha version.

Bye, Max

Max2433BO, 2017-01-22:
Hi, @DarwinNE 

I got an idea: it isn't that the fidocadj 0.24.7 alpha version not accept two identical names of components in the same library section?

Out of curiosity I tried to change the name of each of the "Resistenza" and "Resistenza variabile" duplication (I added a B down)...

... and, as you can see, actually, so the "holes" problem disappears.

[image]

Bye, Max

DarwinNE, 2017-02-05:
Hello,
  thanks, I could replicate the issue. Now I just have to try to understand why it happens :-)

D.

DarwinNE, 2017-02-05:
I think I solved the problem. Thanks @Max2433BO your suggestion was correct, it was related to the name of the component. That greatly helped me and the solution was then easy to find!
Thumbs up!

I am closing this for the moment.
d.

closed 2017-02-05 by DarwinNE