FidoCadJ/FidoCadJ #137: Print out of scale with 0.24.6

lucalcin, 2017-04-26:
Hi,

With the version 0.24.6 was introduced the layer selection for printing. Unfortunately with the same version the scale of the printed PCB is uncorrect, and the image is shrinked more or less by a 10% making the print unusable. 
The same circuit printed with the version 0.24.3 is correct. 
I noticed this bug with OSX Sierra, printing on HP Laserjet.
(The sample image shows two different versions of the same board, but the overall dimension is identical)
[image]

DarwinNE, 2017-04-26:
Hi @lucalcin 
thanks for the info. Someone described something similar on Sourceforge (in Italian): 

https://sourceforge.net/p/fidocadj/discussion/997486/thread/3e9a9343/?limit=25#5a62

I think @Max2433BO did some research.
In any case, is the problem manifesting itself in the same way with version 0.24.5?

https://github.com/DarwinNE/FidoCadJ/releases/tag/v0.24.5

changes about the printing code appeared between 0.24.5 and 0.24.6, but the size calculation is supposed to remain the same.

Cheers,
D.

DarwinNE, 2017-04-26:
I tried to print this:

~~~~
[FIDOCAD]
RV 0 0 1000 1000 0
~~~~

It should be a 12.7 cm x 12.7 cm square and indeed the output size is wrong.
This is a major bug which must be corrected as soon as possible!

Cheers,
D.

lucalcin, 2017-04-26:
Hi again,

anyway even in all previous version I’ve seen that the print was a little bit smaller. With small boards the difference is minimal, but can be a problem with huge board.

Luca

DarwinNE, 2017-04-26:
There was indeed a (quite embarrassing) error in the code I wrote.

Sorry! It should be OK now.

commit 77c3fd86 (2017-04-26): Fix #137
closed 2017-04-26 by DarwinNE