FidoCadJ/FidoCadJ #139: Ver. 0.24.7 makes mess inside an user library

Luigi-C, 2017-08-14:
Version 0.24.7 (Java on Windows 7 64-bit) sometimes makes mess inside an user library when I add one or more new symbols.
Sometimes symbols already present are overwritten by the new ones.

DarwinNE, 2017-08-16:
Hi @Luigi-C 
Thank you very much for the report.

Can you please describe step by step how I can replicate the bug starting from a fresh download of FidoCadJ?
I need that so that I can work on the problem and (hopefully) solve it.

Cheers,
D.

Luigi-C, 2017-08-17:
The following refers to the Italian version of FidoCADJ 0.24.7.
1) Open a new FidoCADJ window
2) Draw a new symbol
3) Select the symbol (Ctrl-A)
4) Mouse right click, select Converti in simbolo
5) Nome file libreria: [select the library to be updated]
6) Gruppo: [select the library group]
7) Nome: add the new symbol name
8) Chiave: [leave unchanged]
9) Check Blocca origine su griglia
10) Click OK
11) If other new symbols have to be added, press Ctrl-A, Del, then repeat steps from (2).

Luigi-C, 2017-08-17:
Please bear in mind that one of the following events may occur, in an unpredictable way:
- The libray is updated correctly;
- The new symbol is not correctly added to the library;
- The new symbol is added, but one or more existing symbols are partially or totally overwritten.

DarwinNE, 2017-08-23:
Thanks again, @Luigi-C, I have some questions though...

Have you notice something strange during that process? I mean, when the third situation occurs ("The new symbol is added, but one or more existing symbols are partially or totally overwritten.") is the software using a particular value for the key? Or a particular name for the component?
Are you using some strange characters somewhere?
Have you tried to have a look at the library files in a text editor?
If you do a Ctrl-Z after you saw FidoCadJ doing something strange, are you able to recover correctly the libraries you are working on?

Cheers,
D.

Luigi-C, 2017-08-23:
I don't force the key value, I leave the default one. It is a numeric value, last time I remember a 3-digit value like 235.
I use alphanumeric chars and sometimes space, left and right brackets in the component name.
After the library was updated I used FidoCADJ and realized that some components were missing from the library; I compared the library with a backup copy and saw that some components definitions have been overwritten, other moved in other library parts.

Luigi-C, 2017-08-23:
Now when I have to add some new symbols to my own library, I use a temporary library, containing just a test text:
1) I create the new symbol
2) I add it to the temp library
3) I open both libraries in a text editor and move the new symbol from the temp library, at the desired position.

DarwinNE, 2020-04-19:
I'd like to solve this issue, but I could not replicate the bug :-(

Luigi-C, 2020-04-20:
Try adding new components to the "libreria aggiuntiva" I have created 
and shared with you. You should make a copy of the library before adding 
components.

After added components, compare the backup copy and the new library, 
checking for possible "mess" (most probably loss of text lines of 
pre-existing components).

This oddly behaviour does not happen always.

DarwinNE, 2020-04-20:
Dear @Luigi-C 
  I will do that.

Can you please put the "libreria aggiuntiva" in a zip file and include here in this discussion?

Kind regards,
D.

DarwinNE, 2020-04-20:
I think I understood something.
Your library is done by editing the file by hand, isn't it?
The format of the library does not specify a certain order for the symbols and for the categories.

When FidoCadJ writes the file, the order of the categories and the symbols inside of the categories is radically different from yours.

For the moment, I only obtained valid files from FidoCadJ, I am trying to see if there are situations where symbols are lost.

Kind regards,
D.

Luigi-C, 2020-04-20:
> it here in this discussion?

Luigi-C, 2020-04-22:
What should I do to convert my lib_agg to a true FidoCadJ-compatible 
version without losing components?

Best regards,
Luigi

DarwinNE, 2020-04-22:
Dear Luigi,
   there is no need to convert the file. FidoCadJ should interpret it (as long it is correct) and when it needs to update it (for example if a new componenti is added), issue another version containing all the components. The order of the components will be changed, though.

Let's consider the following situation:
1 - you start FidoCadJ that shall be configured to use your hand-written library
2 - you define and add a new component.
3 - you realize that another component is missing.

I have the following questions:
- can you please copy/paste here the code of the new component?
- what is the missing component?
- can you please attach here the resulting library (that I suppose will contain the new component and missing another).

Thanks in advance,

Kind regards,
D.

Max2433BO, 2020-05-06:
Hi @DarwinNE 

I tried to add a component to the library as indicated by Luigi,but I was unable to replicate the problem.

If I add a component, the only thing I get is that the library has many components that are no longer sorted, as the original library.

However it does not seem a problem because FidoCadJ shows all of them correctly.

Bye, Max

Luigi-C, 2020-05-06:
Could this behaviour depend (also) on the Java version?

Luigi

⁣Inviato da BlueMail ​