FidoCadJ/FidoCadJ #144: Suggestion: Release a component with the mouse key

Max2433BO, 2018-05-01:
Hi @DarwinNE ,

I open this issue as a remainder for a suggestion proposed by a user of Electroyou site.

In particular, it is asked whether it is possible to release a selected component by pressing the right mouse key, as an alternative to using the keyboard spacebar.

That's all.

Bye, Max

DarwinNE, 2018-08-02:
Hi and thanks for the suggestion.
I am not sure I understand. If I am in the "selection" mode, by clicking the right mouse key, the software activates the contextual menu. In other modes, it depends but if you click directly on the top of the selected element you can set its characteristics.
Should this be modified?

Kind regards,

D.

Max2433BO, 2018-08-12:
sorry I explained myself badly...

Now I try again:

When you select a component and position it on the grid, the same remains attached to the cursor (component highlighted in green).
Now, to unhook the component (green) from the cursor, press the spacebar.

The request was: could it not be that in this case pressing the right mouse button releases the component?

... I hope I have explained myself better with my rough English.

Bye, Max

DarwinNE, 2018-08-12:
Hi @Max2433BO 
thank you for the additional details.
However, there is still something I do not get: if you are editing a drawing and you are in the "edit" mode (i.e. the arrow icon in the toolbox), you are expected to show a contextual menu when you right click. This can be done on top of a circuit element or elsewhere, so I can not use the right click for something else.
Should I add something like "select none" to the contextual menu (there is already "select all")?

Cheers,
D.

DarwinNE, 2018-08-14:
Hi @Max2433BO,
  sorry, but I still do not understand...
When you say that "the selected component attached to the cursor (in drawing mode)" you mean when you are moving an object? But in this case, you have to click and drag the mouse and you are already pressing the left button...
Can you please describe step by step how to get to the situation you are describing?
Maybe I do not understand as I am used to MacOSX and a trackpad and not a real mouse.

Cheers,
D.

Max2433BO, 2018-08-14:
Hi @DarwinNE 

No problem, don't worry:

1) In the component window we choose, for example, a capacitor (In Italian version: Libreria standard -> Componenti discreti -> Condensatore)
2) We point the capacitor with the cursor and, by pressing the left mouse button, select it.
3) Now, if we move the cursor on the grid, a green capacitor will appear that follows all cursor movements
4) If we press the left mouse button, a capacitor is positioned on the grid (appears in black) where the cursor is located, but the green capacitor always remains connected to the cursor and its movements.
5) At present, to disconnect the green capacitor from the cursor, press the spacebar.

The request was: is it possible to implement the disconnection of the component from the cursor (point 5 above) by pressing the right mouse button?

In practice, only in this case, pressing the right mouse button should emulate the spacebar.

By, Max

DarwinNE, 2018-09-25:
Hi @Max2433BO,
  I think now I understand what you mean.

I see what I can do about it.

Cheers,

D.

DarwinNE, 2020-05-01:
The last commit should complete the implementation of this request. I close it.

closed 2020-05-01 by DarwinNE