FidoCadJ/FidoCadJ #145: Thickness Line

Pietro54, 2018-07-31:
Is it possible to implement the thickness of each individual line and not for all?
Using a track is not the same as a thick line.
Best regards
Pietro54

DarwinNE, 2018-08-02:
Hi and thanks for the suggestion.
This has already been discussed in the past (see issue #79). Fractional width for PCB tracks are available since version 0.24.6.
Graphically there are no differences between a "line" and a PCB track with the same width apart the dashing and the arrows. You need them for your drawings?
Kind regards,
D.

Pietro54, 2018-08-03:
Hi Davide

 

The difference between a "line" and a PCB track with the same width is the rounded part that protrudes for the length of the beam.

 

I currently use a solid rectangle when I need a thicker line, but this is more difficult to modify or change  when you need to move the vertices (rectangle cannot rotate with any angle like line, only horizontal or vertical). Even this solution is not perfect on the vertices, see this example:

 

[FIDOCAD]

FJC B 0.5

PL 40 15 150 15 5 0

LI 150 15 150 90 0

EV 170 25 135 5 0

LI 40 45 115 45 0

LI 115 45 115 90 0

EV 130 55 95 35 0

RP 40 65 85 68 0

LI 85 66 85 90 0

EV 91 71 78 61 0

 

I am not able to programme in Java and then cannot help you to modify Fidocadj source. Sorry.

Best regards.

 

Pierino Moro

A: DarwinNE/FidoCadJ
Cc: Pietro54; Author