FidoCadJ/FidoCadJ #146: New "copy as image" needed

DarwinNE, 2018-08-06:
Hi to all,
  I think it would be interesting to add a "copy as image" menu option in FidoCadJ.
This would allow to avoid having to export a drawing into an image format that has then to be imported by other programs such as WYSIWYG text editors etc...

D.

Max2433BO, 2018-09-09:
Hi @DarwinNE ,
I think it's a great idea: which format do you think to use png, jpg or what else?
I believe that PNG and JPG are the most used.

Bye, Max

DarwinNE, 2018-09-11:
Thank you @Max2433BO 

I am not sure about the format, I think the best is to leave the user choose it.
For example, on MacOSX it is much convenient to copy/paste schematics in the PDF format.
The dialog box used for the export can possibly be exploited for that, so I am not expecting implementing this would require a huge amount of work.

Cheers,
D.

DarwinNE, 2020-04-26:
I half-implemented this issue and then realized that there were some implementation issues on the Java runtime environment I used that prevented my code to work.
I updated to Java 14, so that I had to solve #150 first (and #149).

DarwinNE, 2020-04-26:
I think in a first moment I am going to implement it so that the export is done only as a bitmap.

DarwinNE, 2020-04-26:
I close this for the moment. It still requires extensive testing!!!

closed 2020-04-26 by DarwinNE