FidoCadJ/FidoCadJ #150: Java 9 - MacOS Mojave Compatibility

gcantoni, 2019-01-10:
Stackrace: `java -jar fidocadj.jar
The Quaqua look and feel is not available
I will continue with the basic Apple l&f
Exception in thread "AWT-EventQueue-0" java.lang.NoClassDefFoundError: com/apple/eawt/ApplicationListener
   at java.base/java.lang.ClassLoader.defineClass1(Native Method)
   at java.base/java.lang.ClassLoader.defineClass(ClassLoader.java:1009)
   at java.base/java.security.SecureClassLoader.defineClass(SecureClassLoader.java:174)
   at java.base/jdk.internal.loader.BuiltinClassLoader.defineClass(BuiltinClassLoader.java:801)
   at java.base/jdk.internal.loader.BuiltinClassLoader.findClassOnClassPathOrNull(BuiltinClassLoader.java:699)
   at java.base/jdk.internal.loader.BuiltinClassLoader.loadClassOrNull(BuiltinClassLoader.java:622)
   at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:580)
   at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:190)
   at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:499)
   at java.base/java.lang.Class.forName0(Native Method)
   at java.base/java.lang.Class.forName(Class.java:291)
   at net.sourceforge.fidocadj.CreateSwingInterface.run(FidoMain.java:453)
   at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
   at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
   at java.desktop/java.awt.EventQueue.access$600(EventQueue.java:97)
   at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
   at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
   at java.base/java.security.AccessController.doPrivileged(Native Method)
   at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:87)
   at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
   at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
   at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
   at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
   at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
   at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
   at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.lang.ClassNotFoundException: com.apple.eawt.ApplicationListener
   at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:582)
   at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:190)
   at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:499)
`

DarwinNE, 2019-01-13:
Hi @gcantoni 
thanks. I'll leave the issue open though, as AppleSpecific class must be rewritten to avoid obsolete classes.

Kind regards,
D.

DarwinNE, 2020-04-26:
I think the code in `ADesktopIntegration` completely supersedes the old `AppleSpecific` class that is now removed from the code.
The code is indubitably cleaner as we don't need to use reflection anymore to check if "AppleSpecific"  is present. I hope it allows to have a better integration of FidoCadJ with the desktop environment with Linux and Windows.

On MacOS, I migrated from Quaqua l&f to Violetlib VAqua7, that seems to work relatively well. The configuration of the l&f is not complete at this moment, so I opened the issue #177 for this.

I close this issue, but don't hesitate to test the software on Linux and Windows and re-open it (or open more specific issues) if any problem appear.

closed 2020-04-26 by DarwinNE