FidoCadJ/FidoCadJ #154: FidoCADJ-to-OpenSCAD export SVGs for 3D extrusion

paserra, 2019-04-30:
I propose to export selected layers as sequential SVGs (one SVG-per-Layer) and add to FidoCAD a dialog window for helping the user in the creation of an OpenSCAD script for 3D extrusion of vectors.
In this manner you can use the power of FidoCADJ in the creation of stacked layers of 2D vectors and the powerful functions of OpenSCAD in the reconstructions of 3D objects.
3D objects can be obtained by importSVG-extrusion-offset of vector layers but also by using OpenSCAD boolean operators for subtracting extruded layers (at the same offset level).
The obtained 3D objects can be exported as STL and eventually printed.

DarwinNE, 2019-05-02:
This issue is related to a discussion (in italian) here: 

https://sourceforge.net/p/fidocadj/discussion/997486/thread/a2c3e11ab5/?limit=25#3800

Exporting layers as sequential SVGs should be easy to do.
I may allow to create a simple OpenSCAD script, too.

DarwinNE, 2019-05-08:
I decided that it would be interesting to allow to separate layers into different files for all vector formats.

DarwinNE, 2019-05-08:
I think I will implement an export option for that in the command line.
For example: 

~~~~
-m       if a file export is done towards a vector graphic file format,
         split the layers and write one file for each layer. The file name
         will be obtained by appending _ followed by the layer number
         to the specified file name. For example, the following command
         will create files test_0.svg, test_1.svg ... from the drawing 
         contained in test.fcd:

         java -jar fidocadj.jar -n -m -c r2 svg test.svg test.fcd
~~~~

paserra, 2019-05-09:
Hi Davide,
this is amazing!
I wait for you next jar!

DarwinNE, 2019-05-31:
Implementation of this feature request should now be complete. I close it, please reopen if problems are found in versions >= 0.24.8 beta

closed 2019-05-31 by DarwinNE