FidoCadJ/FidoCadJ #165: Windows 10: changing grid size not possible

dewi-ny-je, 2019-11-10:
On windows 10 I tried to change the grid from 5 to 2.5 mm.
All the values appear with dot as decimal separator, and if I just click OK it works.

When I change the grid spacing from 5 to 2.5 I get the message "I cannot read one of the numbers entered". Same message when I use the comma as decimal separator. There may be an issue in the conversion of the decimals.

The system, outside FidocadJ, is set to use comma for decimals.

DarwinNE, 2019-11-10:
Well, the problem is that coordinates in FidoCadJ are always integer numbers. Therefore, a snap grid can not be non integer.

I thought about increasing the resolution, but I fear the incompatibility with older versions.

I am slowly pacing towards introducing non-integer quantities here and there, but this does not come without problems.

dewi-ny-je, 2019-11-11:
As workaround, a clearer error message would do. I really didn't know what the problem was.

DarwinNE, 2019-11-13:
I agree!

commit be0d006b (2020-04-24): Fix #165.
closed 2020-04-24 by DarwinNE