FidoCadJ/FidoCadJ #168: highlight the point of coordinates 100x100

Max2433BO, 2020-01-06:
Hi @DarwinNE ,

I have a question for you:

Is it possible to highlight the point on the 100x100 coordinate grid (with a cross or with a different color), since it is the center for the creation of new components?

Thanks!

Max

DarwinNE, 2020-01-06:
Hello,
  when you define a new symbol in the library, you can put the origin (i.e. the point that becomes (100, 100) in the symbol definition) where you want:

[image]

There is no need to highlight the (100,100) coordinate in FidoCadJ.

Cheers,
D.

Max2433BO, 2020-01-06:
My mistake, sorry!!! o.O

I edited a library file directly, referring to chapter 3.6 of the Italian FidoCadJ manual, where reference was made explicitly to the 100x100 coordinates ...

 Max

DarwinNE, 2020-01-06:
I think the manual is outdated. I had some trouble keeping the doc up-to date and I decided to concentrate myself only on the english documents.

closed 2020-01-06 by DarwinNE