FidoCadJ/FidoCadJ #171: muovere disegni

A3XES, 2020-03-21:
ciao,quando provo a muovere un disegno fatto spesso e volentieri non melo muove ma mi riparte la selezione.e una cosa normale o funziona cosi?e se funziona cosi perche?
io seleziono il disegno premendo il pulsante di selezione nell'interfaccia,tiro un rettangolo verde sopra il disegno che voglio muovere e poi ci clicco sopra,nel 90% dei casi non lo muove ma mi riapre un nuovo rettangolo di selezione.
saluti

DarwinNE, 2020-03-22:
Here is an English translation:

"hello, when I try to move a drawing, often I do not move it but the selection starts again. Is that normal or does it work like that? And if it works like that why?
I select the drawing by pressing the selection button in the interface, pull a green rectangle over the drawing I want to move and then click on it, in 90% of cases it does not move but opens a new selection rectangle."

If you want to move the drawing, you should click on one of the selected elements (but not on one of the red handles). Many vector drawing applications work this way, hence I chosen this behavior for FidoCadJ.

closed 2020-03-22 by DarwinNE

A3XES, 2020-05-16:
ciao,altro problema con i disegni.
faccio una linea tratteggiata,fatta di tante piccole linee continue(lo faccio perche' se faccio una linea continua e la imposto come tratteggiata poi se esporto il disegno come foto,nella foto esce come linea continua e non come tratteggiata) e seleziono tutte le linee che compongono la tratteggiatura poi non le posso muovere tutte insieme,ma le devo muovere una ad una.e normale o sono io che non lo so usare?
altre 2 cose:quando provo a spostare un oggetto molto piccolo,anche se clicco sulla parte verde quasi sempre non lo aggancia correttamente.
quando copio e incollo un oggetto,non lo incolla dove glielo chiedo io ma quasi sopra all'oggetto copiato
ciao e grazie

DarwinNE, 2020-05-17:
Per le linee tratteggiate, questa versione (preliminare, da considerarsi instabile e non definitiva) funziona meglio di quella che stai usando ora?

https://github.com/DarwinNE/FidoCadJ/releases/tag/v0.24.8zeta

Dovrebbe essere possibile esportare in PNG e JPG linee tratteggiate senza che diventino continue.