FidoCadJ/FidoCadJ #173: A little issue of a component

iGreybear, 2020-04-20:
Hi DarwinNE,

in the following component "Simboli Elettrotecnica -> Contatti a due o tre vie -> Commut. bipolare due vie tre pos. centr. di apertura", two pins do not reach the point of grid.

Thanks for your help,
Domenico

DarwinNE, 2020-04-20:
I found those symbols, where the bottom contacts are misaligned with respect to the default 5-unit grid:

[image]

I will slightly shorten those. That will yield a minor incompatibility with previous versions of FidoCadJ, but it is worth to correct this.

DarwinNE, 2020-04-20:
My last commit should correct those components. Don't hesitate to comment if you find similar bugs.

closed 2020-04-20 by DarwinNE