FidoCadJ/FidoCadJ #174: No paste command in text window

Max2433BO, 2020-04-20:
Hi @DarwinNE 

I found this problem in FidoCadJ 0.24.7 on linux s.o (Ubuntu 18.04.4) with jre 1.8.0_131:

if select a text in a window and copy with a right mouse button click then try to paste into the FidoCadJ text window, the paste command does not work (the relative window does not open).

See the attached example video (unzip the file):

[Video.zip](https://github.com/DarwinNE/FidoCadJ/files/4504531/Video.zip)

Bye, Max

DarwinNE, 2020-04-20:
Hi @Max2433BO 
   I tried with a shortcut on MacOS (Command+V) and it works.
Is does the same on Linux (Control+V)?

Kind regards,
D.

Max2433BO, 2020-04-21:
Yes, with ctrl+V it works, but using the menù with the right mouse button no (the menù does not appear)...

Max

DarwinNE, 2020-06-13:
A contextual menu will not be implemented. Keyboard shortcuts will be enough there.

closed 2020-06-13 by DarwinNE